Ultimaker/Cura #5794: [4.0.0] GCode has illegal newline

lukehutch, 2019-05-28:
The gcode generated from Cura has illegal linebreaks in the middle of a comments: (two examples shown below)

```
G91 ;relative positioning
G1 E-1 F300 ;retract the filament a bit before lifting the nozzle
to release some of the pressure
G1 Z+0.5 E-5 ;X-20 Y-20 F100 ;move Z up a bit and retract filament even more
G28 X0 ;Y0 ;move X/Y to min endstops
so the head is out of the way
```

**Application Version**

4.0.0

**Platform**

Linux

**Printer**

Anycubic i3 Mega

**Steps to Reproduce**

Generate gcode from STL

**Actual Results**

See above

**Expected results**

Comments should not contain newlines

**Additional Information**

fieldOfView, 2019-05-28:
Looks like somewhere a comma followed by a space is replaced by a newline.
You wouldn't have a Search and Replace postprocessing script active by any chance?

lukehutch, 2019-05-28:
No, I'm just using Cura straight out of the box, freshly downloaded from the website. I only spotted this because I was curious what the generated gcode looked like.

Ghostkeeper, 2019-05-28:
I'm unable to reproduce this issue. This is what those lines look like when I slice something with Anycubic i3 Mega:

```
G91 ;relative positioning
G1 E-1 F300 ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 ;X-20 Y-20 F100 ;move Z up a bit and retract filament even more
G28 X0 ;Y0 ;move X/Y to min endstops, so the head is out of the way
```

Also in the [source code](https://github.com/Ultimaker/Cura/blob/f6627daa49ecde38df115a88c8bfc61c834518e8/resources/definitions/anycubic_i3_mega.def.json#L61) there are no newlines in that position.

Are you sure it's not your editor that displays it like that or something? It has nice pretty newlines at word breaks instead of a hard border at 80 characters.

lukehutch, 2019-05-29:
The linebreaks are definitely there -- I viewed with both gedit and emacs, and they have the breaks in the same place. They are hard linebreaks, not soft linebreaks.

Here are the input STL and the output gcode files.

[partfiles.zip](https://github.com/Ultimaker/Cura/files/3230204/partfiles.zip)

Ghostkeeper, 2019-05-29:
Yeah I sliced that file but it's not producing the newlines for me.

What I need is a project file. Please go to File -> Save... and zip up the .3mf file you get from that. It'll contain all of your settings and post-processing scripts so it should completely reproduce the exact g-code you're getting.

lukehutch, 2019-05-29:
I can't reproduce this on 4.1.0, so maybe this is fixed now. But I can reliably reproduce it on 4.0.0, so I'll attach my project here, in case the problem is not actually fixed, but somehow isn't triggered in 4.1.0.

Here is my project... it looks like a 3mf file is actually a zipfile anyway, so zipping it again doesn't make it much smaller :-)

[project.zip](https://github.com/Ultimaker/Cura/files/3234848/project.zip)

Ghostkeeper, 2019-05-31:
Ah, I see it now. Indeed the problem reproduces in Cura 4.0. This end g-code was [actually in the 4.0 release](https://github.com/Ultimaker/Cura/blob/59b3046c0f04a357593f18434a286832fd072a89/resources/definitions/anycubic_i3_mega.def.json#L61) with the newlines.

Someone else also discovered this and fixed it for us: https://github.com/Ultimaker/Cura/pull/5594

closed 2019-05-31 by Ghostkeeper