Ultimaker/Cura #5796: Factor K

sim1953, 2019-05-28:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

Ghostkeeper, 2019-05-28:
This is expected. Plug-ins installed from pre-4.0 need to be upgraded in order to run in 4.0. Please perform the following steps:
* Click on "Marketplace" in the top right of your screen.
* Scroll down the list to about halfway and click on "Linear Advance Settings".
* Click on the big blue "Update" button. Or "install" if there is no such button.

In version 4.1 we've guided the user through this process a bit better.

closed 2019-05-28 by Ghostkeeper

sim1953, 2019-05-28:
Thanks you 

----- Mail original -----

De: "Ghostkeeper" <notifications@github.com> 
À: "Ultimaker / Cura" <Cura@noreply.github.com> 
Cc: "sim1953" <simon1953@free.fr>, "Auteur" <author@noreply.github .fr> 
Envoyé: Mardi 28 mai 2019 18:38:32 
Objet: Re: [Ultimaker / Cura] Facteur K (# 5796) 

Ceci est prévu. Les plug-ins installés à partir de la version antérieure à 4.0 doivent être mis à niveau pour pouvoir fonctionner en 4.0. Veuillez effectuer les étapes suivantes: 

    * Cliquez sur "Marketplace" en haut à droite de votre écran. 
    * Faites défiler la liste jusqu'à la moitié environ et cliquez sur "Paramètres avancés linéaires". 
    * Cliquez sur le gros bouton bleu "Mettre à jour". Ou "installer" s'il n'y a pas un tel bouton. 

Dans la version 4.1, nous avons un peu mieux guidé l'utilisateur dans ce processus. 
- 
Vous recevez ceci parce que vous avez créé le fil. 
Répondez directement à cet e-mail, visualisez-le sur GitHub ou désactivez le fil .