Ultimaker/Cura #5797: [4.0.0] Infill has beading issue

lukehutch, 2019-05-28:
I noticed significant beading in the infill of some objects as they were getting printed. I thought this must be a problem with the printer. But I just tried printing a part with a thick and wide base, triggering infill over a large area, and I noticed that the beading follows a distinctive linear "X" battern across the base, so the problem is more likely in Cura. Probably the generated gcode causes the print head to pause at the locations of the beads:

[image]

This is the part (.stl file and gcode):

[partfiles.zip](https://github.com/Ultimaker/Cura/files/3228163/partfiles.zip)

**Application Version**

4.0.0

**Platform**

Linux

**Printer**

Anycubic i3 Mega

**Steps to Reproduce**

Print attached part, with 0.3mm layer height, and all other settings at default

**Actual Results**

See image

**Expected results**

**Additional Information**

Ghostkeeper, 2019-05-28:
Cura doesn't normally command the printer to pause. It looks like this is just a travel move:

[image]

This effect is that the material keeps oozing out of the nozzle while the nozzle travels from one place to another. This oozing is then wiped off by the previously-placed infill lines. You can reduce the effect by tuning some settings:
* Combing Mode to Off. This will cause the printer to retract the material before making this travel move. This takes extra time and wears down the material a bit, but prevents oozing inside your model.
* Z-hop: If retracted, this reduces blobs even more at the cost of some printing time and wear on your Z axis.
* Infill Before Walls: Disabling this causes the travel moves to be made before the infill is placed down on this layer. It could still hit the infill from lower layers but the effect will be reduced. May compromise dimensional accuracy.

closed 2019-05-28 by Ghostkeeper

lukehutch, 2019-05-28:
OK, thanks for explaining!