Ultimaker/Cura #5800: How do I stop Cura from scanning for serial ports?

W0EB, 2019-05-28:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

W0EB, 2019-05-28:
I am an amateur radio operator and when Cura is running - even though I have no printer connected (I print only from SD card) it keeps scanning the serial ports - since my radio equipment IS connected to a serial port, this constant scanning totally screws up the radio equipment's settings.  How can I set the serial port to OFF and get this to stop?  It could damage the radio if it causes it to enter TRANSMIT mode when I'm not watching it.

smartavionics, 2019-05-28:
Hi @W0EB , if you don't actually need Cura to drive a printer using a USB port you can simply disable the USB plugin (do this via the marketplace dialog, yeah, I know really obvious). If you do that, Cura should no longer dick with the USB ports. Hope this helps.

W0EB, 2019-05-28:
Thank you!!!  That fixed the situation completely.  Boy you're right, 
totally obvious NOT - LOL.  I'd never have found that one.

Again, THANKS!

W0EB

serial ports? (#5800)

W0EB, 2019-05-29:
Thanks to smartavionics the solution was found and this is no longer a problem for me.

closed 2019-05-29 by W0EB

eboston, 2021-05-16:
What is the name of the plugin to look for in the Marketplace?  I don't see anything standing out to disable the scan.

jgarder, 2022-04-28:
This is Still necessary for 5.0.0 Beta.. because i had the same issue with arduino projects aswell as with VSCode... anything on any com port really.

ubiquities, 2023-01-23:
For future troubleshooters:

- Open Marketplace
- Click on the Gear icon
- About halfway down find USB Printing
- Click disable, then close down Cura and reopen
- Profit