Ultimaker/Cura #5801: Cura 4.1 Prints very slow Tevo Tornado

TerryLSmith, 2019-05-28:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

diegopradogesto, 2019-05-29:
We would need a little bit more information

Ghostkeeper, 2019-05-29:
Yeah some information was hidden away in the `` comments that are there for the template. I edited the post so that they are visible.

Ghostkeeper, 2019-05-29:
Could you provide a project file that reproduces the problem? You can create a project file in Cura by going to File -> Save...
You'd need to zip up the file to be able to upload it here on Github.

chiefcw, 2019-05-29:
I found that if the maximum z-axis speed is left at zero Cura sends G1 F17987547480000 to Marlin on z-moves which resulted in the z-axis moving extremely slow.  When changed to 5 in Cura it then sent G1 F300 in the gcode.

TerryLSmith, 2019-05-30:
Worked like a charm thanks.

closed 2019-05-30 by TerryLSmith