Ultimaker/Cura #5802: Cura 4.0 slow and not slicing

voodooherbalist, 2019-05-28:
Cura 4.0 is taking forever to open projects and when it does open it gives me an error for slicing....
Slicing engine for Cura has stopped working
A problem caused the program to stop working correctly. Windows will close the program and notify you if a solution is available..

**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

nallath, 2019-05-29:
Without any logging, there is nothing we can do to help you. Please fill in the entire bug report, it's there for a reason.

Ghostkeeper, 2019-05-29:
Could be that problem with scanning for USB ports blocking the job queue?

closed 2020-06-20 by no-response[bot]