Ultimaker/Cura #5803: Tevo Tornado stops printing

Waggy1, 2019-05-29:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

Waggy1, 2019-05-29:
I have just moved up to 4.1 and find that any project I save with 4.1 will hang on my Tevo Tornado stops printing as it builds the foundation of the item I want to print.  I am also surprised there is to undo in 4.1.  I had to go to the help to file ctl Z

misterxp, 2019-05-29:
My printer is home made. Use Marlin as firmware. Just stops printing at random point at start. Tried various, previously made gcodes (made with 4.0) no problem. re-installed 4.1 few times and no change. Created gcode few times no change

smartavionics, 2019-05-29:
Probably same problem as #5804 ?

misterxp, 2019-05-29:
Hello Mark. Thanks for reply. Could be but I removed new Cura and installed old one. Also cancelled gcodes. I will install again and check. Will let you know later tonight/ tomorrow. Thanks ☺

bfcase, 2019-05-29:
My CR-10S does similar, it doesn't actually stop but posses for about 3 or 4 seconds it seems like every time it Z-hops, but I have not verified it is the Z-hop causing the pause.

Ghostkeeper, 2019-05-29:
Duplicate of https://github.com/Ultimaker/Cura/issues/5753

closed 2019-05-29 by Ghostkeeper

misterxp, 2019-05-29:
I will check tomorrow but seems strange it stops at different points and not always at the same point. I.e. First z hop. It just stops and does not continue even after 10 minutes. Heaters are on till I srop the print. Will check mire closely though

Ghostkeeper, 2019-05-29:
It could be that the big number actually causes a buffer overflow attack in your firmware, overriding other memory in the printer itself which then causes arbitrary behaviour.

bfcase, 2019-05-30:
Setting the max Z speed (did I remember that right) from zero to a value seemed to resolve the issue I set my to 20mm/sec

Brian

misterxp, 2019-05-30:
OK will look into it thanks