Ultimaker/Cura #5804: Z hop broken

2kingsrock, 2019-05-29:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

code that causes a stall for 30 seconds.
G1 X50.203 Y52.229 E73.2483
G1 X50.635 Y51.793 E73.28122
G1 F1800 E69.28122
G1 F17987547480000 Z1.3
;MESH:Key Fob v3.0.stl
G0 F3600 X63.082 Y59.766 Z1.3
;TYPE:WALL-INNER
G1 F17987547480000 Z0.3
G1 F1800 E73.28122
G1 F2400 X63.863 Y59.198 E73.33301
G1 X64.431 Y58.417 E73.3848
G1 X64.729 Y57.497 E73.43667
G1 X64.729 Y56.531 E73.48848

same code without z hop (unchecked) , no delay during print
G1 X50.203 Y52.229 E73.2483
G1 X50.635 Y51.793 E73.28122
G1 F1800 E69.28122
;MESH:Key Fob v3.0.stl
G0 F3600 X54.633 Y53.991
G0 X60.291 Y57.1
G0 X62.283 Y58.196
G0 X63.082 Y59.766
;TYPE:WALL-INNER
G1 F1800 E73.28122
G1 F2400 X63.863 Y59.198 E73.33301
G1 X64.431 Y58.417 E73.3848
G1 X64.729 Y57.497 E73.43667

smartavionics, 2019-05-29:
Hello @2kingsrock , I think z-hop is broken by https://github.com/Ultimaker/Cura/commit/ce9f3c9bcc79fdd596bd3c5ac87ce9ba7120a031.

You can work around this by either setting the Maximum Speed Z in the printer definition or by setting or the Maximum Z Speed value in the profile.﻿

trouch, 2019-05-29:
`G1 F17987547480000 Z1.3`
`G1 F17987547480000 Z0.3`

I'm pretty sure most of firmwares cannot parse such high speed value, printer not returning OK then.
The 30s delay sounds to be some host or printer timeout before proceeding with following command.

bfcase, 2019-05-29:
my Cr-10s has this same issue with the 4.1 version . 4.0 worked fine.

Ghostkeeper, 2019-05-29:
Duplicate of https://github.com/Ultimaker/Cura/issues/5753

closed 2019-05-29 by Ghostkeeper

misterxp, 2019-05-29:
I confirm I have same lines in gcode. Took out hop and printer wirked ok and finished the print

2kingsrock, 2019-05-29:
I am running Marlin 1.1.9. using Mega 2560 with RAMPS 1.6 PLUS.
Z hop is something I run often and don't recall 4.0 or previous generating issues when in use.  I can easily setup additional values for the Z to keep this from occurring.  I am surprised to see this large value causing an issue.  I would have thought this would have been ignored since it is larger than the values defined in firmware?  The printer actually seems to act 'stunned' when it processes this part of the gcode.  anyone know if this 'light speed' z value is too large and due to some diff or something it is unable to parse the line?  I think it does not actually z hop (which tells me it didnt finish processing the line), it hangs for some time 10 -30 sec, moves to next location without running a hop and continues to repeat at each hop.  
Fantastic part of this is it just sits melting a hole into the layers underneath, before moving on.    :) 

sample from the configuration.h firmware  
/**
 * Default Max Feed Rate (mm/s)
 * Override with M203
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_MAX_FEEDRATE          { 200, 200, 25, 40 }
/**
 * Default Max Acceleration (change/s) change = mm/s
 * (Maximum start speed for accelerated moves)
 * Override with M201
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_MAX_ACCELERATION      { 3000, 3000, 200, 10000 }
/**
 * Default Acceleration (change/s) change = mm/s
 * Override with M204
 *
 *   M204 P    Acceleration
 *   M204 R    Retract Acceleration
 *   M204 T    Travel Acceleration
 */
#define DEFAULT_ACCELERATION          3000    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  3000    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   3000    // X, Y, Z acceleration for travel (non printing) moves
/**
 * Default Jerk (mm/s)
 * Override with M205 X Y Z E
 *
 * "Jerk" specifies the minimum speed change that requires acceleration.
 * When changing speed and direction, if the difference is less than the
 * value set here, it may happen instantaneously.
 */
#define DEFAULT_XJERK                 10.0
#define DEFAULT_YJERK                 10.0
#define DEFAULT_ZJERK                  0.3
#define DEFAULT_EJERK                  5.0

2kingsrock, 2019-05-30:
I know this has been closed.  I think this is what is occurring.  By default the max z speed is 0 in the 4.1 app.  The pop out says if this is left at '0', the firmware values will be used.  However, leaving this at 0 and checking z hop will insert light speed values for the F value.  This number is so large (F17987547480000), that when the Marlin firmware running on the AT MEGA using the c library strtod causes parsing to fail that move.  This leaves the machine unable to use the defined firmware values, causing the stalls or stops as some people are reporting.  

The simple solution is to force a value for the Z max in the speeds, and not let it be 0.  I set it to 25 mm/s and now the F is reporting 1500 which is a proper value with Z hop enabled.

I think it would be helpful to not use this funny value F17987547480000, or put some of these details in the pop up.  I hope this info helps the next person with printer stops/stalls with Z hop enabled.

[image]

misterxp, 2019-05-30:
Fix works! Thanks ☺