Ultimaker/Cura #5805: Cura 4.1 has strange gcode pause

LeleKaos, 2019-05-29:
**Application Version**
4.1

**Platform**
Windows 10 pro V 1903 Build 18362.145

**Printer**

Creality Ender 3 [MyPrinter.3mf.zip](https://github.com/Ultimaker/Cura/files/3231748/MyPrinter.3mf.zip)

**Steps to Reproduce**

1.  Insert a STL model in Cura 4.1 
2. Slice it whith any parameters configuration 
3. Export it trough file or directly on an external memory or print it via usb

**Actual Results**

The print starts flawlessly with the purge extrusion outside the printing area, starts printing the skirt or brim or even the raft and, when starts to print the actual object, it's starts to pausing after any G0 code on the list and the steppers starts making a rumbling sound, and this is not correct since it is going to ruin the print and also the print bed.

**Expected results**

Normal print

**Additional Information**

smartavionics, 2019-05-29:
Could possibly be the same issue as #5804?

LeleKaos, 2019-05-29:
> Could possibly be the same issue as #5804?

Maybe, let my try without Z hop enabled

LeleKaos, 2019-05-29:
@smartavionics It was exactly the same problem, then I can remove this issue right?

smartavionics, 2019-05-29:
Yes, you can close this issue.

closed 2019-05-29 by LeleKaos

Ghostkeeper, 2019-05-29:
Duplicate of https://github.com/Ultimaker/Cura/issues/5753.