Ultimaker/Cura #5810: [4.1] It doesn't slice

Liger0, 2019-05-29:
**Application Version**
4.1

**Platform**
Win10x64

**Printer**
Cr10 S

**Steps to Reproduce**
Try to load my project and slice

**Actual Results**
The slicing bar goes to 1/4 and then it resets to 0, and it stays stuck there.

**Expected results**
It should slice of course.

**Additional Information**
[CCR10S_test-dog.zip](https://github.com/Ultimaker/Cura/files/3234492/CCR10S_test-dog.zip)

smartavionics, 2019-05-29:
I looked into the crash and this diff stops it occurring. I don't know if it's the best way to fix the problem so I'm just offering this as a starting point for the Cura devs.

````
diff --git a/src/settings/Settings.cpp b/src/settings/Settings.cpp
index 35424125..9d6927ef 100644
--- a/src/settings/Settings.cpp
+++ b/src/settings/Settings.cpp
@@ -96,6 +96,10 @@ template<> ExtruderTrain& Settings::get<ExtruderTrain&>(const std::string& key)
     {
         extruder_nr = get<size_t>("extruder_nr");
     }
+    if (extruder_nr >= Application::getInstance().current_slice->scene.extruders.size())
+    {
+        extruder_nr = 0;
+    }
     return Application::getInstance().current_slice->scene.extruders[extruder_nr];
 }
````

Kinder-Aus, 2019-05-29:
My dual extruding machine does not slice when only using one extruder. If I turn off support and adhesion so that I only use one extruder it will not slice. If I then only turn on support when none will be built it will still freeze while slicing. If Support is required or I turn on adhesion it will slice normally. The issue seems to affect single extrusion as a whole.

For reference I am using:
**Application Version**
4.1
**Platform**
Win10x64
**Printer**
Ultimaker 3 Extended

Liger0, 2019-05-29:
> I looked into the crash and this diff stops it occurring. I don't know if it's the best way to fix the problem so I'm just offering this as a starting point for the Cura devs.
>
> ```
> diff --git [...]

Which crash are you referring to? It doesn't crash, I can even stop the slicing, it just doesn't finish, or even start.

smartavionics, 2019-05-30:
It's the back end (CuraEngine) that is crashing. The GUI just sits there forever waiting for it to complete.

Ghostkeeper, 2019-05-31:
There is an underlying problem though. Somehow the build plate adhesion extruder was set to the second extruder even though there is only one extruder.

Ghostkeeper, 2019-05-31:
How did you create that 0.16LHBrass profile? It has settings in it that you are not able to change in the interface.

Liger0, 2019-05-31:
> 
> 
> How did you create that 0.16LHBrass profile? It has settings in it that you are not able to change in the interface.

From cura with the machine settings plugin.
Also I noticed sometimes the brim use both the extruders (the second touching the model), while sometimes it only uses one extruder.

Ghostkeeper, 2019-05-31:
How can the brim sometimes use both extruders? There is only one extruder in your printer, right?

Liger0, 2019-05-31:
> 
> 
> How can the brim sometimes use both extruders? There is only one extruder in your printer, right?

I am referring to when I use a dual extruder machine with that profile

Liger0, 2019-06-18:
Any update on this? It'd be nice to fix it for 4.2 :\

Ghostkeeper, 2019-06-18:
It's crashing because somehow the adhesion extruder is set to extruder 2, but there is only 1 extruder. Do you know how the adhesion extruder could end up on extruder 2? We are not able to reproduce that on a different profile apart from modifying the profile files.

Liger0, 2019-06-18:
As far as I do remember I created that profile on a cr10s and then imported it into a dual extrusion machine. The second extruder is automatically set to be used in the internal part of the brim, while the external part of the brim goes to the first extruder. So the brim is actually printed with both extruders.

Ghostkeeper, 2019-06-20:
> The second extruder is automatically set to be used in the internal part of the brim, while the external part of the brim goes to the first extruder. So the brim is actually printed with both extruders.

This is not possible unless we have different terminology of what a brim is.

Liger0, 2019-06-20:
Why isn't it possible? It is exactly how it acts.
Inner brim uses second extruder, external brim uses first extruder.
[image]

Ghostkeeper, 2019-06-21:
Ah, no I see what you mean. The brim is printed in yellow there but it puts a skirt around the brim for every extruder.

Liger0, 2019-06-21:
Is it expected to? Because depending on the brim settings and the model it doesn't always put that "skirt"

Ghostkeeper, 2019-06-26:
Yes, it's expected to. It is supposed to make sure that every extruder gets primed on the first layer. Only the extruder that prints the brim itself doesn't need to have an additional loop or anything.