Ultimaker/Cura #5813: Manual Supports not working as expected in 4.1

OldGaf, 2019-05-29:
In Cura 4.0 I am able to have _ONLY_ manually added supports created.
This does not work in 4.1.
[image]

**Application Version**
4.1.0

**Platform**
Win7

**Printer**
CR10S

**Steps to Reproduce**
Un-check Supports, manual add supports and slice/preview

**Actual Results**
No supports are created.

**Expected results**
Manually added supports should be created.

**Additional Information**
This works in 4.0

Ghostkeeper, 2019-05-31:
For me it doesn't even print if Enable Support is properly checked. Seems broken altogether.

OldGaf, 2019-06-03:
Confirmed. 
I did not try to print as I saw them nor displayed in the preview. 
I created a test print and let it print the first layer to check, no manual support created at all.

Italque, 2019-06-29:
Hi, same problem here. The support will be printed only if checked generate supports. If I use some print as supports model they won’t be printed. Useful working  feature in previous versions. If anybody knows how to solve it, please keep me posted about. Thanks

seltix5, 2019-07-18:
Yes it's not printing supports, only if they are touching the bed

Ghostkeeper, 2019-07-19:
Oops, yeah we fixed this in version 4.2, but forgot to close the ticket here. Try the beta, see what you think!

closed 2019-07-19 by Ghostkeeper