Ultimaker/Cura #5814: Stop and go printing Creality CR-10 (4.1 Darwin)

PrinterMan17, 2019-05-29:
**Application Version**
 4.1.0 Darwin

**Platform**
 Mac

**Printer**
 CR-10 Unable to attach file, bug effects all prints 

**Steps to Reproduce**
 Import mesh> slice> print

**Actual Results**
 print stops for a minute  and resumes like normal, then repeats randomly at different times during print

**Expected results**
 should print like normal without stopping until print is over

**Additional Information**
 used both USB cord and SD card, happens regardless, never had this issue until 4.1.0, have cleaned printer and hot end with same results, have restarted printer multiple times, happens with seemingly all files I try and print all with the same outcome

smartavionics, 2019-05-29:
Same as #5753?

PrinterMan17, 2019-05-29:
No, thing to do with feed rates, printer just pauses randomly

2kingsrock, 2019-05-30:
Do you have Z hop enabled?  Check your GCode for this 'G1 F17987547480000'  Is this the same issue as this #5804?

PrinterMan17, 2019-05-30:
That GCode is present in the file, but I've noticed the print pausing on the same layer

PrinterMan17, 2019-05-30:
Im going to go ahead and close this issue out, changing the Max Z speed to 25 seemed to of fixed the problem. Ive ran a 50 min print and all seems to be well. Thank you!

closed 2019-05-30 by PrinterMan17