Ultimaker/Cura #5817: Crash on sign in

bvalat, 2019-05-30:
**Application Version**
4.0.0 and 4.1.0

**Platform**
WIN 10 64

**Printer**
Ender 3

**Steps to Reproduce**
Open Cura
Click on sign in
Enter your credentials

**Actual Results**
Cura instantly crash and leave browser with a localhost url and empty page.

**Expected results**
Signed in without crashed or at least an error message.

**Additional Information**
Click on sign in opens this url on my browser : 
https://account.ultimaker.com/authorize?code_challenge_method=S512&redirect_uri=http%3A%2F%2Flocalhost%3A32118%2Fcallback&scope=account.user.read+drive.backup.read+drive.backup.write+packages.download+packages.rating.read+packages.rating.write+connect.cluster.read+connect.cluster.write+cura.printjob.read+cura.printjob.write+cura.mesh.read+cura.mesh.write&state=%28.Y.%29&code_challenge=GW8olkDjalzrUbBUR4JeDpRI9yiVPZ1ovQTqMhvdxDztbuZNvl_O0nxpz5yiYWt1Hrlr4imODApXA-Q3mSgy8A%3D%3D&response_type=code&client_id=um----------------------------ultimaker_cura

When cura crashed the left url is 
http://localhost:32118/callback?code=aDlVH133O6iZLkn5Hw1TiwAf_gv3aSeq&state=%28.Y.%29

Tried Edge (clean)  and Chrome (many plugins) as main browser.

bvalat, 2019-05-30:
Here is a complete logfile from start to crash.
[cura.log](https://github.com/Ultimaker/Cura/files/3237610/cura.log)

ChrisTerBeke, 2019-05-31:
Mm your log does not contain any relevant warnings or errors for this problem. Maybe your computer is blocking Cura from setting up a small webserver (localhost) to receive the authorization callback on?

bvalat, 2019-05-31:
My computer is just with windows defender (no fancy antivirus software)....
I don't see why the embded weserver will fail and crash without any warning the whole software.
Will try with antivirus/firewall disabled tonight to confirm.

Ghostkeeper, 2019-05-31:
If it crashed without logging anything or leaving an error message, then that's an indication that something got a segfault (or similar signal) in the front-end.

bvalat, 2019-06-09:
Tried with windows defender off (both real time protection and firewall).
Still crashing.
Do you have a debug version for traping the  embded webserver crash?

Ghostkeeper, 2019-06-10:
No, we don't have that. At least, not for CPython or Qt which is probably where the bug resides. One thing we could try is to run Cura's front-end through GDB: [instructions for a different application also apply to Cura.](http://rpg.hamsterrepublic.com/ohrrpgce/GDB_on_Windows) That is really technical though.

We're unable to reproduce this bug so far, even when interrupting network connectivity at various points in the process.

gsxr2001, 2019-08-01:
Identical problem!! Some news?

bvalat, 2019-08-05:
Still have the problem with 4.2.1.
Tried gdb. The tools refuse to load the exe : not in executable format: File format not recognized.  Will try running from source.

bvalat, 2019-08-11:
Sign in works from source... 
Get some errors but works:
`2019-08-11 14:12:52,967 - INFO - [MainThread] cura.Machines.QualityManager.getQualityChangesGroups [179]: Cannot find node for machine def [creality_base] in QualityChanges lookup table
2019-08-11 14:12:54,820 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [188]: Error check finished, result = False, time = 22.8s
2019-08-11 14:13:02,424 - DEBUG - [MainThread] cura.AutoSave._onTimeout [56]: Autosaving preferences, instances and profiles
2019-08-11 14:13:05,058 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.startAuthorizationFlow [146]: Starting new OAuth2 flow...
2019-08-11 14:13:05,553 - DEBUG - [MainThread] cura.OAuth2.LocalAuthorizationServer.start [49]: Starting local web server to handle authorization callback on port 32118
127.0.0.1 - - [11/Aug/2019 14:13:07] "GET /callback?code=TMSC_LNhijJkdQMqUt1YoiA2umnhMHJD&state=%28.Y.%29 HTTP/1.1" 302 -
2019-08-11 14:13:07,695 - DEBUG - [Thread-11] cura.OAuth2.AuthorizationService._storeAuthData [206]: Attempting to store the auth data
2019-08-11 14:13:08,148 - DEBUG - [Thread-11] cura.OAuth2.LocalAuthorizationServer.stop [63]: Stopping local oauth2 web server...
Exception in thread Thread-11:
Traceback (most recent call last):
  File "C:\Users\valat\AppData\Local\Programs\Python\Python35-32\lib\threading.py", line 914, in _bootstrap_inner
    self.run()
  File "C:\Users\valat\AppData\Local\Programs\Python\Python35-32\lib\threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\valat\AppData\Local\Programs\Python\Python35-32\lib\socketserver.py", line 232, in serve_forever
    ready = selector.select(poll_interval)
  File "C:\Users\valat\AppData\Local\Programs\Python\Python35-32\lib\selectors.py", line 323, in select
    r, w, _ = self._select(self._readers, self._writers, [], timeout)
  File "C:\Users\valat\AppData\Local\Programs\Python\Python35-32\lib\selectors.py", line 314, in _select
    r, w, x = select.select(r, w, w, timeout)
OSError: [WinError 10038] Une opération a été tentée sur autre chose qu\u2019un socket

2019-08-11 14:13:18,149 - DEBUG - [MainThread] cura.AutoSave._onTimeout [56]: Autosaving preferences, instances and profiles`

Ghostkeeper, 2019-08-11:
For the part of the world that is not yet conquered by the French, WinError 10038 says "an operation was attempted on something that is not a socket".

It seems to happen during the closing of the oauth2 web server. Looking at [this similar issue](https://stackoverflow.com/questions/15210178/python-socket-programming-oserror-winerror-10038-an-operation-was-attempted-o) we might be writing something to a socket that is already closed. Maybe that's causing the crash.

nallath, 2019-08-12:
I've added a bit of code that should resolve the OS error that you got.

Ghostkeeper, 2019-08-12:
That's all we can really do from this side without being able to reproduce it. The log-in should still succeed because it only closes the socket after having completed the handshaking. It's not a perfect solution, but I think this is all we're going to be able to achieve towards this bug so we'll consider it completed then.

closed 2019-08-12 by Ghostkeeper

gsxr2001, 2019-08-12:
Sorry, I didn't understand which is the solution. There is not?

Ghostkeeper, 2019-08-12:
Yes, it was fixed here: https://github.com/Ultimaker/Cura/commit/ce2b1adc825875f71fce8d2d3ca64c49c9d17a2e

We can't reproduce the error so we can't really validate it well, but we choose to make Cura ignore this error and continue with what it was doing for the rest. In theory that should be fine.

bvalat, 2019-08-12:
Yes! It is a python SSL bug in 3.5.2
Any ssl operation crash silently on some of my computers.
I replaced _ssl.pyd by the 3.5.4 one and everything works.
See #6152