Ultimaker/Cura #5818: Ender 3 CURA 4.1.0

davekh94, 2019-05-30:
**Application Version**

CURA ULTIMAKER 4.1.0 - does not occur in CURA ULTIMAKER 4.0.0
**Platform**

MS WS 10 PRO
**Printer**

ENDER 3
**Steps to Reproduce**

start a print?
**Actual Results**

printing head stop mid print, thinks, continues, leaves unnecessary dots.
**Expected results**

it should print continously?
**Additional Information**

smartavionics, 2019-05-30:
Same issue as #5753.

akiims, 2019-05-30:
Ender 3, Cura 4.1.0
If I use z-hop, it happens super slow. It takes ~10 sec to move 1 mm up and again ~10 sec to lower the nozzle.

Ghostkeeper, 2019-05-31:
Possibly, but the reporter is not very specific.

If you set the Maximum Z Speed setting to 10 (up from 0), does that fix the issue?

frzdaframe, 2019-06-01:
I'm having the exact same problem. I've now tested an successful print sliced before and after updating. One prior to updating still prints, no issues, print sliced after updating exhibits same symptoms. After every movement, stops for way to long. Atleast 4-5 seconds. Moves to start of next movement, stops, same delay, and so on. Turned a 5.5 hour print i started last night, into a 13 hour and 52 minute print. 

Going to test, moving back to previous build and see if result is any different.

fieldOfView, 2019-06-01:
@frzdaframe perhaps if you have the exact same problem, the exact same workaround as mentioned by @Ghostkeeper in the post above you would have worked?

frzdaframe, 2019-06-01:
@fieldOfView Honestly i don't doubt it would have. I didn't give it a shot. To be honest, it was just nice to confirm it wasn't the PrinterMods hotend LED (why would that even right?) causing the issue. I had just finished a nice, no issues print, installed the LED, then go to start up the next print, not thinking i had updated CURA a few days earlier, and have this issue. So seeing that i wasn't the only one, made me think of the update, and i was already testing different things before i found this post. 

So i just rushed to revert, and issue fixed. I know this is a site more for programmers, and people who are much more advanced than me haha, but i figured I'd share. I'm only like 4 months new to the 3d printing community so, look at me go. :"D

davekh94, 2019-06-01:
Guys, I really appreciate all your comments and help but I have moved back to cura 4.0 and the problem does not occur anymore. I am pretty new to 3d printing so I would not play around with things I do not fully understand. 
The problematic process was the followong:
1. Start a print
2. Print head stops mid print (most likely at dotty spots or columns but did exhibit the problem throughout just a simple line) 
3. Thinks on the spot for 5 to 30 seconds. Moves a tiny bit. Thinks again. 
4. Repeats this until he thinks that particular spot is finished. (yep, if he feels like it, he does that single spot for 10 minutes) 
5. Eventually he gets things done, but obviously standing at the same spot for 10 minutes causes not very nice plastic balls. The super tiny movement barely prints anything, but the nozzle heat causes PLA to flow anyway. 
6. Same print in CURA 4.0 goes smoothly without any problems. 

I hope I was specific enough.

fieldOfView, 2019-06-01:
> if there's anything i can do to help the issue

Yes, try the workaround, and report if it fixes the issue. In all likelyness it does, but it could be that there is another issue. If you don't let us know, we will never know. 

Contraray to what seems to be popular belief, reporting that you reverted Cura to a previous version does not get a problem fixed. Neither does posting "me too", or reiterating the problem when a workaround has already been posted.

frzdaframe, 2019-06-01:
@fieldOfView Fair enough, and i completely get what your saying. As i said, i was just relieved to know i wasn't having an electrical issue after installing that LED. My follow up comments weren't intended to frustrate you. I understand the process of troubleshooting things I'm familiar with, hence my detail about what i did/didn't do. Granted, after my response to your initial response to me, i went back and read @Ghostkeeper possible resolution, and with my limited knowledge being so new to cura and my ender 3, i can't say i know how to do any of that at this point, so i just figured, let me revert for now, and let the "SMEs" handle it.

But i will gladly give it go tomorrow, and see if i can attempt those settings recommended, and give some feedback on whether it resolved the problem on my Ender 3 Pro or not. Nice to see your passion about getting things like this fixed though. ;)

TomHersey, 2019-06-02:
Ender 3 - 4.0.0 vs 4.1.0   Same issue here...I even side my side compared the gcode files in notepad and matching configurations in the program, the gcodes ended up varying right from the start, within 10 lines, they rearranged the initial configuration files and added extra lines of code,  By the time the code got past the just the adhesion raft I used, the lines were already 8 lines off just in programming length, and the coordinates programmed in stopped lining up immediately.  
Not a programmer per se, but I know some simple data comparisons and basic coding and this new software is off.  I was using to differnt PCs to verify the difference making sure I didn't mess up the printer(i was about to dismantle the extruder arm and test the motors for issues, glad I didn't.

END RESULT---REVERT to 4.0.0

Ghostkeeper, 2019-06-03:
> Thinks on the spot for 5 to 30 seconds. Moves a tiny bit. Thinks again.

Yeah, thanks. That is an indication to me that this is indeed a duplicate of #5753 (rather than the Marlin buffer underrun problem). I'll close this as duplicate then.

Please try the workaround to set the Maximum Z Speed to 10mm/s until we publish a fix.

closed 2019-06-03 by Ghostkeeper