Ultimaker/Cura #5819: [4.0.0] Head crash after changing Infill Density

msquaredct, 2019-05-30:
I print the same piece over and over.  All I did was change the Infill Density from 30% to 20%.

Then print head crashed into the piece the next three  attempts.  It crashed at three different layers, not the same one over and over.

I put the setting back to 30% and it's printing without error.

**Application Version**

4.0.0
**Platform**

Windows 7
**Printer**

Creality-CR-10S
**Steps to Reproduce**

Changed only Infill Density, nothing else.
**Actual Results**

Head crashes into piece.
**Expected results**
Should print without error.
**Additional Information**

smartavionics, 2019-05-30:
Hello @msquaredct , please save the project file (File -> Save) and either rename the .3mf file to .zip or zip it and attach to this issue. Without that, it's unlikely we can help.

msquaredct, 2019-05-30:
[CCR10S_Infill20-Fails.zip](https://github.com/Ultimaker/Cura/files/3238341/CCR10S_Infill20-Fails.zip)
[CCR10S_Infill30-Works.zip](https://github.com/Ultimaker/Cura/files/3238343/CCR10S_Infill30-Works.zip)

smartavionics, 2019-05-30:
I think the problem is caused by a combination of settings that make the nozzle travel across the infill a lot and because you are not using retraction with z-hop the nozzle can catch a high point on the infill.

The infill pattern you are using will cause problems because on each layer the infill crosses and so where the infill lines cross, it will be high. You need to use an infill pattern that doesn't cross. Try using the gyroid infill pattern, it doesn't cross and also you can get away with low infill densities. For your model I would be looking at maybe 10% gyroid infill.

Other changes that will make a difference are that you should choose the z-seam alignment to be shortest rather than sharpest corner as that will cut down on the amount of travel over the infill.

Another setting that is generally worth enabling is the optimize wall printing order although for this particular model it won't make much difference.

Hope this helps.

[image]

Ghostkeeper, 2019-05-31:
The general solution that Cura provides for this problem is to turn Combing off and enable Z-hop. This comes at the cost of some printing time and wear on your printer's Z axis though.

closed 2019-05-31 by Ghostkeeper