Ultimaker/Cura #5823: Cura 4.1 only starts "as administrator"

BigLebowski59, 2019-05-31:
**Application Version**
4.1

**Platform**
Windows10

**Printer**
don´t get to that point

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**
The Cura GUI doesn´t come up, when I start Cura normally. The cura process is there in the task manager, but the GUI doesn´t come up. There are also no logs written, no cura.cfg created.
When I choose "Start as Administrator" from the Windows start menu, the Cura GUI comes up and I can work like normal.

Ghostkeeper, 2019-05-31:
I'm unsure exactly how this works in Windows 10, but sometimes when a file was written by an administrator account it cannot be overwritten by a non-administrator account. Perhaps Cura has been started once as administrator and now it can't write anything any more.

That would mean that if you set the access rights to the `C:\Users\<BigLebowski59>\AppData\Roaming\cura` folder and its subfolder correctly, it might get fixed.

Alternative theories: Maybe your home directory is unusual?

nilrog, 2019-06-04:
I have the same problem...never executed Cura as Administrator. It did work when I initially downloaded it afair, but not now. Was forced to install Windows updates this morning and now I can't launch it...but I cannot say that it was working before that either since I did not try.

cura.log contains nothing from today...the last entry is from 29/5...which is when I downloaded it and launched it for the first time. There should be no permissions problem because files have been created/updated at the time when I tried to launch it again.

Ghostkeeper, 2019-06-04:
Could you try running the application at `C:\Program Files\Ultimaker Cura 4.1\CuraCLI.exe`, and see what happens to the log file then?

BigLebowski59, 2019-06-04:
nothing, the logs stay empty and only a empty command window comes up

Btw, in the meantime I can´t start cura as admin anymore. I have to run 
it in Win8 compatibility mode. That wroks (for now...)

<author@noreply.github.com>
Gesendet: 04.06.2019 14:53:42
(#5823)

Ghostkeeper, 2019-06-04:
> I have to run it in Win8 compatibility mode. That wroks (for now...)

Oh, so it must be something that Windows changed between Win8 and Win10 I suppose :worried:

Maybe it's something with the supported OpenGL version in your video driver? Maybe the Windows 8 compatibility mode uses a different video driver?

BigLebowski59, 2019-06-04:
I don´t know, funny enough Cura 4.0 worked for a while and Cura 3.6 
worked all the time

Gesendet: 04.06.2019 15:23:44
(#5823)

Ghostkeeper, 2019-06-04:
We didn't change anything significantly in the rendering engine or in the GUI framework between 3.5 and now.

closed 2020-06-20 by no-response[bot]