Ultimaker/Cura #5824: Switch beetween Custom Selection and other Views reset's Custom Selection View

xenon2008, 2019-05-31:
**Application Version**

4.1

**Platform**

**Printer**

Creality Ender 3

**Steps to Reproduce**

Switch between Views.
If i have selected my own custom visible Options and switch once to a different View like Basic/Expert/Advanced and then Back to Custom Selection, my previously selected Options (whitch  have selected in the Options..) are gone....
After i switched once to an another View, Custom Selection does not show any more my custom selected Options.
This works fine in previous Versions...

**Actual Results**

**Expected results**

**Additional Information**

diegopradogesto, 2019-05-31:
I can't reproduce it.

I can only see this behaviour if I select "Show All Settings". But this is expected, since this option is equivalent to "Custom Selection" + selecting all settings.

xenon2008, 2019-06-01:
Hi!

I now have tested it again... Selected all my Settings which i want to see under "Custom Selection"
After i switched once to "Basic" and then Back to Custom Setting my Settings are lost..

In 3.6 and Cura 4.0 has this worked without Problems.
Any Ideas what i could do?

fieldOfView, 2019-06-01:
As a workaround, there’s a plugin in the Marketplace that lets you save your custom selection into a new set.

Ghostkeeper, 2019-06-03:
I also couldn't reproduce this bug. Here's a video showing what I did: http://dulek.net/work/custom_setting_stays_visible.mkv

After switching back to Custom, the Enable Ironing setting remains visible.

Is there something you are doing different? What exactly are you clicking on?

xenon2008, 2019-06-04:
Good Morning!

I have now tested a few Things, and i think i have found the problem!

it looks like the error comes from the "Printer Settings" plugin. I'll be watching it at the weekend or the next few days again and test it and then give you a second notice!

Thank you and best regards!

Ghostkeeper, 2019-06-04:
If you've confirmed that it comes from there, you may want to report the issue here: https://github.com/fieldOfView/Cura-PrinterSettingsPlugin/issues

Vandrasc, 2019-07-19:
I think we can close this one now that we established what the issue is. 

Thank you for your input @xenon2008 !

closed 2019-07-19 by Vandrasc