Ultimaker/Cura #5825: 4.1 Creality CR-10S random pauses

zipspence, 2019-05-31:
**Application Version** 4.1 (happened with 4.0 also)

**Platform** Win10, i7-7700K, Nvidia 1070's in SLI Driver 430.86.

**Printer** CR-10S, issue happens with all projects. 

**Actual Results** The printer randomly pauses during the print, seems to happen more at the end/start of a layer. 

**Expected results** The printer will not stop or pause during the print. 

**Additional Information** I am using the Marlin from TH3D version R2.11
I also have the EZABL and various other upgrades.

smartavionics, 2019-05-31:
Most likely same as #5753.

zipspence, 2019-05-31:
testing various settings now.

Seref, 2019-05-31:
Same here!
**Application Version 4.1** (didn't happen with 4.0)
**Platform** Win10
**Printer** Modified/Custom 3D Printer, issue happens with all projects at the start on Layder 1&2

I upgraded to 4.1 and the first thing that happened was that everything just stopped after some certain point.
Doesn't matter if I print over USB or export it onto a SD Card and print it from there, the printer just stops (it doesn't freeze since the you can still access the menu and stop the print).

zipspence, 2019-05-31:
Issue has been corrected, default value on Maximum Z Speed was the issue, *input 60 mm/s* 

Case resolved, thankyou  for the help smartavionics for the tip. 

I had to read further into the case #5753 in order to see the resolution

zipspence, 2019-05-31:
Issue was resolved set max z speed

closed 2019-05-31 by zipspence

xavierator, 2021-02-03:
Hi, I think I have a similar problem, but difficult to understand for me.

I just purchase the printer 1 week ago, I finished right 2 prints and after that all prints stop randomly.

I'm using CURA 4.8.0.

What dou you think ?

Ned I to change the maximum Z speed in printer settings, upgrade the printer firmware, change something in CURA, ... ?

Many thanks, Xavi

zipspence, 2021-02-03:
Go into cura and go into your speed options. You will need to expand the
extra options for this menu in your setting and find your Z axis speed
setting and set it to a reasonable value. 150 or so.

xavierator, 2021-02-03:
I just expanded all speed options:

[image]

But I can't find Z axis speed:

[image]

Thanks

Ghostkeeper, 2021-02-10:
Maximum Speed Z is a machine setting and not visible in the interface. It can normally only be changed by printer definitions.
There is a plug-in in the marketplace called Printer Settings which will make those settings visible. This then allows you to change settings which Cura considers to be inherent to the printer or its firmware.

In Cura 4.8, the Maximum Speed Z is only used as maximum for setting values, not as default, unless you've modded Cura. So changing it would not have an effect on the print, just on the time estimates and on how far you can adjust settings that relate to Z movement speed (like Z hop speed and wire printing speeds).

zipspence, 2021-02-10:
Z hop speed

xavierator, 2021-02-10:
**Mistery discovered:** in my case is a filament sensor problem, I'll need to change the filament sensor that detect wrong the end of filament when there are filament.
Thanks for your help