Ultimaker/Cura #5826: Support "oozy" materials in multimaterial prints

fughilli, 2019-06-01:
**Is your feature request related to a problem? Please describe.**
When slicing for multimaterial prints on multi-nozzle machines (all nozzles on the same gantry), it seems that Cura assumes that the materials being printed are not significantly "oozy", in the sense that if a nozzle transitions to print temperature, it will not leak a significant amount of material while the active nozzle is working. My observation is that this is not true for flexibles, and especially for flexibles in a bowden system. Slicing a multimaterial print with TPU/PLA using Cura 4.1.0 (see attached config/model files) results in a machine crash, since the TPU nozzle begins to ramp up to temperature while the PLA nozzle is still working, and oozes blobs of TPU all over the part and build area, which can catch the other nozzles. This results in skipped steps, blemishes/warts on the final part, etc. Likewise, since this oozed material leaves a void in the nozzle, a large extra_prime value is required (which must also be configured to the expected maximum ooze, which varies depending on how long the nozzle was idle at temperature and other factors).

**Describe the solution you'd like**
To solve this problem, I propose a per-material setting which indicates the "ooziness" of the material. This could be a binary value ("oozy" or "not oozy"). When checked, the following behaviors should be observed:

1) The oozy material nozzle will not ramp temperature until it is the active tool.
2) If a prime tower is configured (which it should be...):
    2a) during temperature ramp, the nozzle will trace the priming path (without E-steps)
    2b) Once the nozzle has reached temperature, the nozzle will retrace the priming path (with E-steps)
    2c) After the layer(s) using the oozy material have been completed, the nozzle will retrace the priming path during temperature ramp-down (without E-steps)
3) If a prime tower is not configured, these wipe-during-ramp moves could be performed over the part toolpath (perhaps optionally the infill?)

This ensures that any material that escapes the nozzle during temperature ramp will be wiped continuously onto the priming tower. The wipe feedrate should be chosen such that the nozzle reaches the standby temperature before finishing the wipe movement (or the nozzle could perform multiple passes over the wipe path). Since the nozzle is also blocked by cooled material during the wipe procedure, ooze is minimized, and a lower (or zero) extra_prime amount can be employed.

The "oozy" setting should be per-material, so as to not preclude overlapping temperature ramps for materials that do not behave this way. For example, on my TPU/PLA prints, the PLA does not ooze significantly, and the wiping during ramping is superfluous. As such, the PLA nozzle could ramp to temperature while the TPU nozzle is finishing its wipe/ramp-down.

**Describe alternatives you've considered**
Not really an alternative, but the interior space of the prime tower could be used to discard blobs instead of wiping during temperature ramp-up. I didn't like this idea because for N>2 materials, inter-material adhesion might be too low for these blobs to stick and stay inside the tower.

**Affected users and/or printers**
Anyone doing true multi-material printing on a multiple-nozzle single-gantry machine.

**Additional context**
I have written a [post-processing script that demonstrates the general idea](https://github.com/fughilli/multimaterial-cura-postprocess/tree/temp_by_extrude). See the attached zip file for a pre/post gcode sample of a dual material PLA/TPU wheel (STLs and the Cura project file are also included).
[cura_feature_request_files.zip](https://github.com/Ultimaker/Cura/files/3243427/cura_feature_request_files.zip)

Ghostkeeper, 2019-06-03:
I'll discuss the possibility of disabling the pre-heat option with the team.

I didn't try your post-processing script, but I am curious how you managed to implement what you describe, with a continuous wipe until the temperature has been reached. Did you predict how long it will take and generate enough lines to fill that time, or did you find a way to make movements during an M109 command or something of the like?

fughilli, 2019-06-03:
I made an external measurement of the thermal velocity of the nozzle and used that to adjust the feedrate for the duplicated prime tower segments so that the nozzle would be (roughly) at temperature before continuing. To be safe, I inserted an `M104` before the wipe sequence and an `M109` after to enforce the correct temperature after the ramp. This has the unfortunate side effect of inserting a small delay even when the nozzle is at the correct temperature (presumably while Marlin makes multiple measurements). I think reusing `Preheat::getTimeToGoFromTempToTemp` internally and just `M104` would get this to work just fine.

The postprocessing script makes a fair amount of assumptions about the particular hardware, but I think the critical pieces to have here to keep the machine from crashing are the wipe-during-ramp and making the ramp only occur while the tool is active.

khumarahn, 2019-06-20:
I am very intetested in this. Trying to print with two PETGs of contrast colors, the prints are a mess because of oozing from inactive nozzle. I have to print at 245 degrees, and have to cool down to 180 or so to stop oozing. Cannot do it without postprocessing currently.

fughilli, 2019-09-06:
I have a proof of concept patch attached above. @Ghostkeeper could you provide some pointers on finishing the implementation as described in the commit log?