Ultimaker/Cura #5827: [4.0.0] Preview color scheme is not distinct enough

protist, 2019-06-02:
**Application Version**
4.0.0

**Platform**
Arch Linux

**Printer**
N/A

**Actual Results**
Select `Preview` > `Color Scheme` > `Line type`.

Two colours in particular are difficult to tell apart: `Top / Bottom` is yellow, and `Infill` is orange.

**Expected results**
It should be easy to tell them apart. There are quite a few unused colours, so using more distinct colours would be helpful. Alternatively, the colours could be customisable.

**Additional Information**
I've picked a simple area for the screenshot, but with more complex models, it can be difficult to discern infill from top/bottom.
[image]

Ghostkeeper, 2019-06-03:
I agree with this actually. Bit of a long-standing issue that has always been a bit of a papercut.

Ghostkeeper, 2019-06-03:
Similarly with support and support interface.

Ghostkeeper, 2019-06-07:
[image]
Better.

closed 2019-06-07 by Ghostkeeper

protist, 2019-06-07:
That looks great; thank you!

commit ea1b8eef (2019-06-07): Give infill and support interface a more distinct colour

We had complaints that infill had a too similar colour to skin.

Fixes #5827.
commit ec22ddcb (2019-06-07): Give infill and support interface a more distinct colour

We had complaints that infill had a too similar colour to skin.

Fixes #5827.