Ultimaker/Cura #5830: [Feature Request] Printer name in Preview Mode

yet-another-average-joe, 2019-06-02:
**Is your feature request related to a problem? Please describe.**

Printer name is not displayed in Preview Mode

**Describe the solution you'd like**

Please, display current printer name in Preview Mode

**Describe alternatives you've considered**

display current printer name in Preview Mode

**Affected users and/or printers **
 
**Additional context**

The printer name not being displayed in Preview Mode is somewhat confusing as the "Print with Octoprint" button also does not tell where it will send the file.

Ghostkeeper, 2019-06-03:
Good point. Let's give this to our UX designer.

What do you think would be a good place to display it?

yet-another-average-joe, 2019-06-03:
The layout could be like so, the printer combobox being on the left 

I just realize that the material also is not displayed in preview mode. There's probably room for it. 

[image]

[image]

[image]

The combo could be beside the Cura logo in the tab ribbon. But it would not be very logical...

GregValiant, 2024-11-24:
I'm cleaning house.
Using the "Sidebar GUI" plugin from the Marketplace rearranges the workspace so that the information you are asking for is displayed.
Can this be closed?

closed 2024-12-08 by github-actions[bot]