Ultimaker/Cura #5831: [4.1] Visible seam is generated in the middle of flat wall

extesy, 2019-06-02:
**Application Version**
4.1

**Platform**
Windows 10 x64

**Printer**
Wanhao i3 Duplicator Mark II

**Actual Results**
Travel points converge in the single point in the middle of the wall which generates a visible seam when printed:
[image]
Z-seam preferences (also in screenshot) are set to hide seam in the sharpest corner.

**Expected results**
Travel points should converge into a corner instead.

**Additional Information**
Model: [wither.3mf.zip](https://github.com/Ultimaker/Cura/files/3245017/wither.3mf.zip)

smartavionics, 2019-06-02:
> Travel points converge in the single point in the middle of the wall which generates a visible seam when printed:

OK, so your objects do have a vertex part way along what othewise looks like a straight wall. That's in the model. By choosing to use the sharpest corner z-seam style, Cura may choose those vertices as the z-seam position. You can stop it doing that by instead selecting the user specified z-seam style. The sharpest corner style rarely does what one wants and, I believe, is best avoided.

[image]

extesy, 2019-06-02:
Why would Cura choose those vertices if there are better "sharper" options available, like the actual corners of the model?

smartavionics, 2019-06-02:
Hi, this is part of your model mesh, you can see it has a vertex on the seemingly straight wall...

[image]

If you zoom right in, you can see that there's a little kink in the wall and one of those corners must be < 90 deg and so it wins the prize as the sharpest corner. This is one of the reasons why the sharpest corner is not really a very good criteria for choosing the z-seam

[image]

Ghostkeeper, 2019-06-03:
There is no real solution for subjective cases like this, where a human can understand that the corner is too small to be considered a corner but the computer will identify the corner there like any other. We'll consider this to be a modelling error rather than a problem with Cura.

Setting the Z Seam Alignment to User Specified by default is not an option unless we have some way to set the default for the X,Y position as well depending on the model. Otherwise the prints will still have the same issue but with different prints.

In this case you can expect the corner to still be visible on the print though even when the seam is somewhere else, because the nozzle will slow down to make this corner.

closed 2019-06-03 by Ghostkeeper