Ultimaker/Cura #5833: Filament Color change and stepper motors

fiddlekrazy, 2019-06-02:
**Application Version**
Cura 4.0/4.1

**Platform**Windows 10

**Printer**

**Actual Results**

**Expected results**

**Additional Information**

smartavionics, 2019-06-02:
Hello @fiddlekrazy , please zip up an example of gcode that has this problem and attach to this issue. Thanks.

fiddlekrazy, 2019-06-03:
The problem is not in the gcode but must be in the post processing plug-in. As you can see, in the gcode attached, there is nothing that either enables or disables the stepper motors. 

[Eye.zip](https://github.com/Ultimaker/Cura/files/3245668/Eye.zip)

smartavionics, 2019-06-03:
I wonder if your controller has an idle timeout and if you delay when changing the filament, the timeout cuts the power to the steppers?

Ghostkeeper, 2019-06-03:
Cura tends to assume that the motors remain powered throughout the whole print unless Cura commands it so. For instance, it commands it explicitly to disable power to the motors during a pause for Repetier (and then homes the print head when restarting).

What extra g-code does your printer need to prevent it from disabling power?

fiddlekrazy, 2019-06-08:
Cura must be commanding the stepper motors to disable. Ender 3 uses Marlin. M600 is the only code that Cura sends to the machine so the disable code must be embedded in the M600 command in Marlin. Does this make sense??

Ghostkeeper, 2019-06-10:
`M600` by itself doesn't have such an option according to [this documentation](https://reprap.org/wiki/G-code#M600:_Filament_change_pause). There are commands to [enable/disable power](https://reprap.org/wiki/G-code#M17:_Enable.2FPower_all_stepper_motors) to the motors but since it appears that `M600` in itself disables it and doesn't execute any further g-code until the user resumes, it is impossible to add a command in between to re-enable power or something.

If there is nothing that Cura can send to prevent this for your printer, you're really going to have to look towards the firmware developers of that printer. `M600` appears to have some flaws in their implementation.

fiddlekrazy, 2019-06-11:
I can understand that it disables the extruder motor but I do not understand why it disables the other motors. I guess I will have to try and write a workaround in gcode.

closed 2019-06-11 by fiddlekrazy