Ultimaker/Cura #5835: [Feature Request] Confirmation dialog after clicking "Print with..." button

yet-another-average-joe, 2019-06-02:
(I wrote this FQ a couple of hours ago but it seems I forgot to click Submit... I can't find it, so I repost, hoping it is not a double post !)

**Is your feature request related to a problem? Please describe.**

There's no confirmation dialog when clicking the "Print" button. This can be confusing.

**Describe the solution you'd like**

A confirmation dialog box that could display :

- the information normally displayed by the "?" button
- the printer's name the stl was sliced for and where the gcode file will be sent
- a checkbox with "Automatically start print job after uploading", preset according to the printer parameter in the connect panel ; I don't know about USB as I never connected a printer to a PC using the USB port...
- an "OK" button
- a "Cancel" button

**Describe alternatives you've considered**

**Affected users and/or printers **
 
**Additional context**

- the button "Print with..." is hit just after the model is sliced and parameters are set. A couple of times, I had to run to the printer (in another room) because the printer was preheated, and some stuff was still laying on the buildplate (wrench, cutter or whatever).  Sanity check !

- the parameter "Automatically start print job after uploading" is set on a per printer basis. Why should some printers start automatically while some others won't ? This parameter is "hidden" deep in the interface while it is a really important one

Ghostkeeper, 2019-06-03:
> A couple of times, I had to run to the printer (in another room) because the printer was preheated, and some stuff was still laying on the buildplate (wrench, cutter or whatever). Sanity check !

Well the printer DOES ask you whether or not you've cleared the build plate after a print, so it should assume that you're not lying to it :wink:

fieldOfView, 2019-06-03:
I think OP is using OctoPrint, which has no such question.

yet-another-average-joe, 2019-06-03:
Yes, I'm using Octoprint. There's no confirmation dialog before printing. But even if it had, it is probably interesting to have a summary before clicking "Print". This is usual with speadsheets, word processors and all apps that print.

In the first post, I wrote : "the information normally displayed by the "?" button"
I meant : the information normally displayed by the "i" button

Ghostkeeper, 2019-06-07:
We decided to not implement this. The confirmation for printing with Ultimaker clusters is also found to be annoying (but necessary in order to select the printer to print with).

I think you'll have to resort to using the abort button.

closed 2019-06-07 by Ghostkeeper