Ultimaker/Cura #5837: [4.1] UI and UX really bad for ultrawide monitors

frank26080115, 2019-06-03:
**Application Version**
4.1

**Platform**
Windows 10, monitor is an ultrawide 3440x1440 monitor, I got it to for better CAD work

**Actual Results**

[image]

UI elements way too spread apart, causing disassociation between parent and child UI elements, and unnecessarily long mouse travel

In real world distances, the words "select configuration" and the nozzle drop down are **a foot and a half** away from each other.

**Expected results**
UI elements be grouped together closely so the user can quickly use them and so that sub-elements are not too far from the parent UI element.

Ghostkeeper, 2019-06-03:
Should we maybe left-align that drop-down then? Though that will have some implications for small monitors.

dstulken, 2019-07-12:
I'd suggest adding a dynamic/flexible spacer between the left and middle drop-down menus.  
The left menu would anchor to the left (where it currently is), the right menu would anchor to the right (where it currently is), and the now reasonably sized middle menu would anchor toward the right, keeping its drop-down where it currently is.

It would result in some non-interactive area within the horizontal menu bar, however I would agree with @frank26080115 that the items are so distant right now, that they are disassociated... if anything, having the whole bar active makes a click so far to the left that open a menu on the right seem like a misclick.

fieldOfView, 2019-07-12:
This whole issue is what prompted me to write the Sidebar GUI plugin. Have you tried it?

frank26080115, 2019-07-12:
Yes actually I was disappointed when the upgrade from 4.0 to 4.1 broke your plugin, but glad it's back

However this means I can't upgrade to 4.2 in the future unless I know that Sidebar GUI will keep up with it.

I really appreciate your great work @fieldOfView and I would totally understand if one day you get bored and move on to other hobbies. That's the nature of open source.

So I hope that more REAL changes to the default UI will take place, with real discussions over what is objectively wrong with it.