Ultimaker/Cura #5838: Cura 4.1.0 adds something to model (in the brim area)

OldChris, 2019-06-03:
Short description : Cura adds something to the model in the brim area.

Software : CURA 4.1.0

Operating System
	Windows 10 Pro 64-bit
CPU
	AMD FX-6100	18 °C
	Zambezi 32nm Technology
RAM
	8,00GB Dual-Channel DDR3 @ 666MHz (9-9-9-24)
Motherboard
	MSI 760GM-P23(FX) (MS-7641) (CPU1)	27 °C
Graphics
	hp L2035 (1600x1200@60Hz)
	hp L2035 (1600x1200@60Hz)
	3067MB ATI Radeon 3000 Graphics (MSI)
Storage
	465GB SAMSUNG HD502HJ ATA Device (SATA )	36 °C
Optical Drives
	No optical disk drives detected
Audio
	High Definition Audio Device
Graphics
		Monitor 1
			Name	hp L2035 on ATI Radeon 3000 Graphics
			State	Enabled, Output devices support
			Multiple displays	Extended, Secondary, Enabled
			Monitor Width	1600
			Monitor Height	1200
			Monitor BPP	32 bits per pixel
			Monitor Frequency	60 Hz
			Device	\\.\DISPLAY1\Monitor0
		Monitor 2
			Name	hp L2035 on ATI Radeon 3000 Graphics
			Current Resolution	1067x800 pixels
			Work Resolution	1067x769 pixels
			State	Enabled, Output devices support
			Multiple displays	Extended, Primary, Enabled
			Monitor Width	1600
			Monitor Height	1200
			Monitor BPP	32 bits per pixel
			Monitor Frequency	60 Hz
			Device	\\.\DISPLAY2\Monitor0
		ATI Radeon 3000 Graphics
			Manufacturer	ATI
			Model	Radeon 3000 Graphics
			GPU	RS780
			Device ID	1002-9616
			Subvendor	MSI (1462)
			Current Performance Level	Level 0
			Current GPU Clock	196 MHz
			Current Memory Clock	800 MHz
			Current Shader Clock	800 MHz
			Die Size	64 mm²
			Release Date	2009
			DirectX Support	10.0
			DirectX Shader Model	4.0
			OpenGL Support	2.0
			GPU Clock	350,0 MHz
			Driver version	8.970.100.9001
			ROPs	4
			Shaders	40 unified
			Memory Type	HyperMemory
			Memory	3067 MB
			Pixel Fillrate	1,4 GPixels/s
			Texture Fillrate	1,4 GTexels/s
				Count of performance levels : 2
					Level 1 - "Perf Level 0"
					Level 2 - "Perf Level 1"

**Printer**
Ultimaker 2

**Expected results**
The print is just a box with a hole in it's center.
CURA adds this small attachement.

**Additional Information**
See screenshots and OpenSCAD files, and STL file

best regards, Chris.

[UM2_test.zip](https://github.com/Ultimaker/Cura/files/3247923/UM2_test.zip)

[test_OpenSCAD.zip](https://github.com/Ultimaker/Cura/files/3247929/test_OpenSCAD.zip)
[image]
[image]
[image]
[image]

Ghostkeeper, 2019-06-04:
Hmm, yeah that's strange. There is no extra geometry there or anything. It's printed as support but there is nothing to support there.

Disabling Use Towers removes the effect so it clearly thinks that there is an edge to support there.

OldChris, 2019-06-05:
> 
> Disabling Use Towers removes the effect ....
Thanks for spotting that ! My other PC (laptop) now also shows this appendix.

closed 2024-12-07 by github-actions[bot]