Ultimaker/Cura #5839: Machine Start_gcode customization for Heated chamber

KOUBeMT, 2019-06-03:
Hello there.

Is there a way to modify / override machine start gcode when a a specific material / quality profile is used.

One application would be  the activation of a heated chamber module when ABS plastic is used. This is one application but i'm sure others will use it for other purposes...

We would like to set it globally too, which means if we're in a multi extrusion configuration and at least one material requiring heated chamber is used, the global machine start gcode gets modified / overriden...

thanks for your help.

Ghostkeeper, 2019-06-04:
No, start g-code can't be edited in the material or quality profile because it can't be edited per extruder. There is only one start g-code but you might have multiple extruders in your printer. If two materials or quality profiles say different things about what the start g-code must be there is no way to resolve the conflict.

A material or quality can override the extruder start and end g-code, which is a piece of g-code that gets executed upon switching extruders. That's only for dual extrusion though.

In Cura 4.2 we'll have a new setting called Build Volume Temperature that might be useful for you though. This adds a `M141 S35` command to the g-code which should be interpreted as changing the temperature of the heated chamber to 35 degrees Celcius. If that doesn't work for your firmware you can always implement it yourself in the start g-code by adding something like this:

    M9001 S{build_volume_temperature}

Or whatever M code you're using.

KOUBeMT, 2019-06-04:
That resolves directly my problem. If i can suggest an additional feature that would be useful

Ghostkeeper, 2019-06-04:
I don't think the use case for using `M191` instead of `M141` is very large. Printers that are expensive enough to have a heated chamber will normally also have a heated build plate, and during the first few layers the heated build plate will overpower the heated chamber by orders of magnitude. So if the chamber takes 5 minutes to warm up, that will not be a problem since it's still printing very close to the build plate.

closed 2019-06-04 by KOUBeMT

ETE-Design, 2020-11-06:
@Ghostkeeper I know this is Closed... But is there a way to change Build Volume Temperature from M141 Sxx to M191 Sxx?

Ghostkeeper, 2020-11-17:
@ETE-Design yes there is. You can add a post-processing script "Search and Replace". Make it search the text `M141 S` and replace that with `M191 S`.

ETE-Design, 2020-11-17:
@Ghostkeeper So not just possible to edit a parameter in one of your files instead?

Ghostkeeper, 2020-11-19:
No. Well, unless you can compile CuraEngine yourself.

You'd need to edit this line here:

https://github.com/Ultimaker/CuraEngine/blob/0852f3cd5e909c9ebb459befb499292460da97d4/src/FffGcodeWriter.cpp#L580

To say:

    gcode.writeBuildVolumeTemperatureCommand(mesh_group_settings.get<Temperature>("build_volume_temperature"), true);