Ultimaker/Cura #5841: Top/Bottom Pattern - plugin for adding novel patterns?

david-coneff, 2019-06-03:
**Is your feature request related to a problem? Please describe.**

A problem I notice with an FDM printer making a solid flat surface along the XY-plane is that the default zig-zag pattern of "lines" and "zig zag" both result in a tool-movement pattern of each filament line being extruded directly next to a filament that is still cooling. this results in a differential cooling pattern along the length of the filament that can make it extra susceptible to distortions and curling.

In contrast, the 1st layer of the "Raft" type Build Plate Adhesion sticks very well to the build plate because each strand laid across the build plate is completely separate and does not significantly interact thermally with the previous filament laid down.

**Describe the solution you'd like**

In order to make a well-formed flat face that doesn't curl, a modified "top/bottom" pattern is needed that marries these two ideas -  generates g-code similar to a raft that always draws the outside border first, then connects long straight filaments to this outside border that have intentional space between them, then makes a second pass over the same z-layer by hop-moving and fills in the empty spaces between the straight filament lines covering the top/bottom shell. I think the appropriate name for this would be "interlaced zig zag" top/bottom pattern.

**Describe alternatives you've considered**

Making a 2nd interleaved pass over the same z-layer requires non-trivial changes to the g-code of a complex part more than a few layers deep with varying profiles along the z-axis since it would require hop-moves after every straight line is laid down against a border outline. Code-reuse of the "raft" build plate adhesion generator would be a quick way to implement this, but I don't know how since I'm not familiar with Cura's code base. If there is someone else that could implement it quickly I think it would be of general use to everyone in improving print outcomes with flat, completely sealed defect-free surfaces.

**Affected users and/or printers**

I think this would benefit 3D FDM printers in general to make filling in large flat XY-planes more reliable and curl-free. It takes advantage of physics general to the process of FDM printing and not of a specific printer.

Ghostkeeper, 2019-06-04:
If I may summarise, I think you're requesting an option to change the order in which skin lines are printed (in particular for the first layer) to interweave the lines, so that at least half of the lines stick well to the build plate and the second weaving is only to fill the gaps and make it strong.

I think that warrants an experiment.

We know that if you print lines consecutively this *already happens* due to the printing process, at least with the Ultimaker S5. It just so happens that the first line can flow unobstructed but the second line gets obstructed in its flow by the first line so much that the material stays mostly on top of the first line. Then the third line can flow under the second line again mostly unobstructed. It alternates this way.

david-coneff, 2019-06-06:
Here is a short video that hopefully gives some visual clarification to what I'm requesting:

coneff labs
Ultimaker Cura 4.0 Top/Bottom Patterns - novel access patterns for filled in polygons
(https://www.youtube.com/watch?v=9IJxAyXOcwY)

Ghostkeeper, 2019-06-07:
I think we understood your feature request correctly :) As far as I understood from your explanation, it's meant to improve two things:
* Bed adhesion, as explained before.
* Pillowing, in particular for pure blends of PLA. I'm not convinced that the purity of the material has much to do with it but you never know what weirdnesses plastics have. I think it might just be more visible with translucent materials.

We've now put the feature on our backlog. I can't tell what priority it'll receive.

david-coneff, 2019-06-08:
I can't say that it's specifically purity, but some filaments that are heavier on the filler tend to have a more "flat" bead vs. translucent filaments that tend to form a very round bead that cools quickly. They need higher melt temps to extrude smoothly.

The raft + interlaced fill-in pattern triggering for any top/bottom pattern sections in a given layer with 3+ consecutive filaments next to each other in a closed polygon would be the solution to those problems - bed adhesion, better surface quality (less pillowing), and better water-tightness of an individual layer for both top and bottom layers so that fewer holes occur between strands. 1-2 layers with crossing orientations would make a part reasonably solid.

It looks like I should have put the feature request in the CuraEngine repo instead, but if you have it on the backlog I suppose it's taken care of.

Ghostkeeper, 2019-06-10:
> It looks like I should have put the feature request in the CuraEngine repo instead since the Cura repo is more UI/UX focused, but if you have it on the backlog I suppose it's taken care of. Thanks!

Not really. We generally prefer issues to go into the Cura repository anyway because the bug tracker is used more.

Ellecross, 2019-06-11:
for reference, here's the internal Cura ticket: CURA-6576