Ultimaker/Cura #5842: [4.1.0] Crash after un-grouping scaled part

Hardwater526, 2019-06-03:
**Application Version**
4.1.0
**Platform**
Windows 7
**Printer**
S5
**Steps to Reproduce**
Group two models, scale and un-group them. 
**Actual Results**
Cura crashes with a fatal error
**Expected results**
The models should un-group in their scaled size and orientation.

Ghostkeeper, 2019-06-04:
Yup, I could reproduce this. Pretty bad. This is the stack trace:

```
Traceback (most recent call last):
  File "/home/ruben/Projects/Cura/cura/CuraApplication.py", line 1514, in ungroupSelected
    op.push()
  File "/home/ruben/Projects/Uranium/UM/Operations/Operation.py", line 50, in push
    Application.getInstance().getOperationStack().push(self)
  File "/home/ruben/Projects/Uranium/UM/Operations/OperationStack.py", line 54, in push
    operation.redo()
  File "/home/ruben/Projects/Uranium/UM/Operations/GroupedOperation.py", line 48, in redo
    op.redo()
  File "/home/ruben/Projects/Cura/cura/Operations/SetParentOperation.py", line 28, in redo
    self._set_parent(self._parent)
  File "/home/ruben/Projects/Cura/cura/Operations/SetParentOperation.py", line 57, in _set_parent
    self._node.setParent(new_parent)
  File "/home/ruben/Projects/Uranium/UM/Scene/SceneNode.py", line 272, in setParent
    self._parent.removeChild(self)
  File "/home/ruben/Projects/Uranium/UM/Scene/SceneNode.py", line 384, in removeChild
    self._resetAABB()
  File "/home/ruben/Projects/Uranium/UM/Scene/SceneNode.py", line 701, in _resetAABB
    self.boundingBoxChanged.emit()
  File "/home/ruben/Projects/Uranium/UM/Signal.py", line 212, in emit
    self.__performEmit(*args, **kwargs)
  File "/home/ruben/Projects/Uranium/UM/Signal.py", line 319, in __performEmit
    signal.emit(*args, **kwargs)
  File "/home/ruben/Projects/Uranium/UM/Signal.py", line 212, in emit
    self.__performEmit(*args, **kwargs)
  File "/home/ruben/Projects/Uranium/UM/Signal.py", line 315, in __performEmit
    func(dest, *args, **kwargs)
  File "/home/ruben/Projects/Uranium/UM/Qt/Bindings/ActiveToolProxy.py", line 96, in _onPropertyChanged
    self._updateProperties()
  File "/home/ruben/Projects/Uranium/UM/Qt/Bindings/ActiveToolProxy.py", line 115, in _updateProperties
    self._properties[name] = property_getter()
  File "/home/ruben/Projects/Uranium/UM/../plugins/Tools/ScaleTool/ScaleTool.py", line 230, in getObjectWidth
    return float(Selection.getSelectedObject(0).getBoundingBox().width)
AttributeError: 'NoneType' object has no attribute 'width'
```

Ghostkeeper, 2019-06-04:
You don't even need to scale the models. You just need to have the scale tool selected.

nallath, 2020-06-22:
I've just tried reproducing this, it looks like it has been fixed in the meantime.

closed 2020-06-22 by nallath