Ultimaker/Cura #5843: Upgraded to Cura 4.1. Can see my models normally as before but after slicing, I am only to see them in preview as x-ray view (in blue). Casnnot see anything on the screen (no grid or frame either) under layer view.

hol-tech, 2019-06-04:
**Application Version**

**Platform**

**Printer**

**Actual Results**

**Expected results**

**Additional Information**

Ghostkeeper, 2019-06-04:
This usually means that slicing failed somehow.

Could you share a project file with us? You can save the project file using File -> Save... and then upload it on the Github website in a comment.

hol-tech, 2019-06-05:
[image]
[image]

The two screenshots show that in x-ray view the object can be seen but when going into layer view everything is just white. 
this happens with different models placed.

hol-tech, 2019-06-05:
[image]

here is the model in 'prepare' mode before slicing. Once sliced and clicking onto the 'preview' button it disappears but can be seen in x-ray view.

hol-tech, 2019-06-05:
all i can see in preview window are the sliders on the right and in the bottom.

hol-tech, 2019-06-05:
from seeing the slicing happen (it took around 20 seconds) it looks to me that there is no problem with that but there is a display/ graphics issue at stake...

Ghostkeeper, 2019-06-07:
Hmm, looks like a rendering error to me.

Perhaps you also have a log file for me? It should indicate if you don't have a supported OpenGL version.

You could try going into Preferences and enabling the compatibility mode for layer view. That creates a less nice looking layer view, but using a lower OpenGL version so it will be supported on more computers.

You could also try updating your graphics drivers.

closed 2020-06-20 by no-response[bot]