Ultimaker/Cura #5846: [4.1 & 4.0] Does not support .3mf file type

Block137, 2019-06-04:
**Application Version**
Both 4.0 and 4.1

**Platform**
Windows 10
Acer Predator Helios H300

**Actual Results**
Cura can export but will not open .3mf project file
.3mf is not in the list of supported file type.
[image]

**Expected results**
Prompt asking to import models only or whole project

**Additional Information**
"Open 3MF files with Cura" was checked during install
[image]

Ghostkeeper, 2019-06-04:
Could you supply a log file of Cura? See https://github.com/Ultimaker/Cura#logging-issues

I suspect that it'll say something about the 3MF reader plug-in not being able to load.

I'm unable to reproduce this myself. It could also be platform-dependent.

Block137, 2019-06-04:
> I suspect that it'll say something about the 3MF reader plug-in not being able to load.

I checked Marketplace and re-enabled 3MF reader. Probably disabled by accident since its's on top of the list.
Its's working now

closed 2019-06-04 by Block137