Ultimaker/Cura #5848: Link Variant to Extruder

TiZYX, 2019-06-04:
**Application Version**
4.1.0

**Printer**
TiZYX EVY DUAL

Hi,
Do you know if it is possible to link a variant to an extruder ? I would like the "Classic Extruder" variant be associated with extruder 1 and the "Direct Drive" variant be associated with extruder 2 of the TiZYX EVY DUAL.
That means that when someone add the TiZYX EVY DUAL printer on cura, the 1st extruder is a Classic Extruder and the 2nd is a Direct Drive. Until now, both extruders are by default Classic Extruder.

Thanks

Ghostkeeper, 2019-06-04:
The only thing that can currently be limited to a specific extruder number is the extruder definition. That is, the files `tizyx_evy_dual_extruder_0.def.json` and `tizyx_evy_dual_extruder_1.def.json`. It is not possible to limit the variant to one of the two extruders right now. I see your use case for it, but I don't think that's a big enough use case to build that (as it could get a bit complex). It's not worth our effort.

Changing the default though has more of a use case for us. I'll put that up for my colleagues to consider.

TiZYX, 2019-06-04:
Thanks for your answer.
Maybe we misanderstood. We just want that when we add a TiZYX EVY DUAL printer, the extruder 1 is considered as a "classic extruder" and the extruder 2 is considered as a "Direct Drive".

Ghostkeeper, 2019-06-04:
I think I understood that correctly then. You want it to say "Classic extruder" next to the [1] extruder icon in the top bar (underneath "Generic PLA") and to say "Direct Drive" next to the [2] extruder icon, right?

The text displayed there is indeed the name of the variant, so basically you want the first variant to be "Classic Extruder" and the second variant to be "Direct Drive".

What I was saying is that we're probably not going to build a system that it always says "Classic Extruder" at [1] and "Direct Drive" at [2]. Right now you can already supply both variants to both extruders, which will then appear in a drop-down menu if you click on the middle section of the top bar. However right now it will by default select a certain variant for both extruders (defined in `tizyx_evy_dual.def.json`). I said that we'll consider making it possible to select a different default for each extruder. The user will then still be able to select a different option in the drop-down, but unless the user changes that it will select the correct feeder for each extruder by default.