Ultimaker/Cura #5849: Cura 4.0 and 4.1 with CR10 MINI

rudy1987, 2019-06-04:
Hello, 

I have a big probleme with Cura 4 and my CR10 Mini.

Le speed of extruder is wrong.

With Cura 3.6 and printer parameter : CR10 for my CR10Mini = No probleme
With Cura 4.0 and printer parameter : CR10 for my CR10 Mini = impossible to print
(for print something with CR10Mini and Cura 4.0 : extruder 300% and speed 30% of 60mm/s)

What is the good printer parameter for my CR10Mini on Cura 4 .??

rburema, 2019-06-05:
Could you maybe send us a project file where this happens? (Save, rather than export, then you can attach it to a comment here by renaming it to zip.)

Ghostkeeper, 2019-06-06:
The only speed-related change in between those versions is this one: https://github.com/Ultimaker/Cura/commit/9a9f58428524ee53bfe04b680cc8e1a23cc4f928

If you change the Jerk settings to 20, does it work better for you?

closed 2020-06-20 by no-response[bot]