Ultimaker/Cura #5850: Unwanted support

filinda, 2019-06-04:
**Application Version**

1.4.0

**Platform**

Windows 10

**Printer**

ENDER 3

**Actual Results**

Support appears in place where it should not be

**Expected results**

I dont want to waste material and time on usless support

**Additional Information**

file https://drive.google.com/open?id=1XCtZeuko1l9RTXUc9LfqM36c70pc7iw0
imege with issue https://drive.google.com/open?id=1wy2baNWLyvTaHcLI8cBAEYnc1wzk7SGd

rburema, 2019-06-05:
It's hard to completely prevent/rule out cases like this without compromising somewhere else.

There is a workaround however: You can specify an additional offset from the model with the 'support x/y distance' and 'minimum support x/y distance' parameters.

Ghostkeeper, 2019-06-06:
Duplicate of https://github.com/Ultimaker/Cura/issues/4219.

closed 2019-06-06 by Ghostkeeper