Ultimaker/Cura #5851: [4.1.0] slicer terminated with std::bad_alloc and hangs

timhunderwood, 2019-06-04:
**Application Version**
4.1.0

**Platform**
Windows 10 - 64bit
16GB RAM

**Printer**

**Steps to reproduce**
Added an stl file (mesh with 120k vertices - 1.5MB) and click slice.

**Actual Results**
Cura slicing engine process appears in task manager and begins ramping up in memory and processor usage. It suddenly drops to 0% and the log (see below) shows a:

```terminate called after throwing an instance of 'std::bad_alloc'```

Full log after clicking slice:

```2019-06-04 17:49:20,261 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [225]: Starting to slice...
2019-06-04 17:49:20,261 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [243]: Going to slice build plate [0]!
2019-06-04 17:49:20,601 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [429]: Sending slice message took 0.3389766216278076 seconds
2019-06-04 17:49:20,706 - DEBUG - [Thread-10] UM.Backend.Backend._backendLog [101]: [Backend] [WARNING] Mesh has overlapping faces!
2019-06-04 17:49:20,742 - DEBUG - [Thread-10] UM.Backend.Backend._backendLog [101]: [Backend] [WARNING]  -s magic_fuzzy_skin_point_dist="0.8" -s line_width="0.4" -s support_tree_enable="False" -s connect_skin_polygons="False" -s start_layers_at_same_position="False" -s material_print_temperature="210" -s machine_max_jerk_e="5.0" -s support_line_distance="2.6666666666666665" -s xy_offset_layer_0="0" -s raft_surface_fan_speed="0" -s top_thickness="0.8" -s bottom_thickness="0.8" -s coasting_min_volume="0.8" -s roofing_layer_count="0" -s skin_preshrink="0.8" -s jerk_prime_tower="20" -s support_xy_overrides_z="z_overrides_xy" -s support_conical_min_width="5.0" -s jerk_wall_x="20" -s extruder_prime_pos_x="0" -s retraction_retract_speed="25" -s support_brim_line_count="20" -s switch_extruder_retraction_speed="20" -s support_skip_zag_per_mm="20" -s bridge_skin_speed_2="15.0" -s machine_feeder_wheel_diameter="10.0" -s support_bottom_extruder_nr="0" -s jerk_travel_layer_0="30.0" -s support_enable="False" -s bridge_fan_speed_2="0" -s layer_height="0.2" -s travel_compensate_overlapping_walls_enabled="True" -s wireframe_top_delay="0" -s support_bottom_line_width="0.4" -s bottom_skin_preshrink="0.8" -s adaptive_layer_height_threshold="200.0" -s wipe_hop_enable="True" -s prime_tower_enable="False" -s retraction_hop_enabled="False" -s machine_nozzle_heat_up_speed="2.0" -s support_fan_enable="False" -s retraction_combing_max_distance="0" -s ironing_only_highest_layer="False" -s raft_fan_speed="0" -s meshfix_keep_open_polygons="False" -s acceleration_wall_0="3000" -s acceleration_infill="3000" -s raft_base_line_width="0.8" -s switch_extruder_retraction_speeds="20" -s wall_0_wipe_dist="0.2" -s mesh_position_z="0" -s meshfix_union_all="True" -s machine_nozzle_tip_outer_diameter="1" -s gradual_support_infill_steps="0" -s roofing_line_width="0.4" -s bridge_skin_speed="15.0" -s speed_support_roof="40.0" -s skin_line_width="0.4" -s speed_support_interface="40.0" -s support_interface_enable="False" -s cool_min_layer_time="5" -s wireframe_roof_inset="3" -s cross_support_density_image="" -s machine_max_acceleration_y="9000" -s adaptive_layer_height_enabled="False" -s support_tower_roof_angle="65" -s center_object="False" -s acceleration_topbottom="3000" -s support_z_distance="0.1" -s support_connect_zigzags="True" -s build_volume_temperature="35" -s material_diameter="2.85" -s fill_outline_gaps="False" -s infill_line_width="0.4" -s acceleration_ironing="3000" -s bridge_fan_speed_3="0" -s jerk_travel="30" -s wipe_retraction_prime_speed="25" -s cooling="0" -s material_adhesion_tendency="10" -s support_tree_collision_resolution="0.2" -s spaghetti_flow="20" -s travel_compensate_overlapping_walls_0_enabled="True" -s wipe_hop_speed="100" -s resolution="0" -s infill_pattern="grid" -s infill_wipe_dist="0.1" -s meshfix_extensive_stitching="False" -s adaptive_layer_height_variation="0.1" -s wireframe_flow_connection="100" -s layer_start_y="0.0" -s brim_line_count="20" -s acceleration_support="3000" -s z_seam_y="615" -s machine_nozzle_cool_down_speed="2.0" -s draft_shield_height="10" -s minimum_support_area="0.0" -s jerk_topbottom="20" -s material_shrinkage_percentage="0" -s wireframe_drag_along="0.6" -s cool_min_layer_time_fan_speed_max="10" -s support_bottom_height="1" -s material_flow_dependent_temperature="False" -s mesh_position_y="0" -s material_print_temperature_layer_0="210" -s top_skin_preshrink="0.8" -s extruder_prime_pos_abs="False" -s infill_sparse_thickness="0.2" -s draft_shield_height_limitation="full" -s prime_tower_wipe_enabled="True" -s prime_tower_position_x="195.6" -s support_interface_density="100" -s wireframe_nozzle_clearance="1" -s jerk_print="20" -s bridge_skin_support_threshold="50" -s max_feedrate_z_override="0" -s material_flow="100" -s retraction_speed="25" -s support_roof_height="1" -s bridge_settings_enabled="False" -s support_interface_pattern="concentric" -s infill_sparse_density="20" -s ironing_pattern="zigzag" -s machine_min_cool_heat_time_window="50.0" -s spaghetti_infill_extra_volume="0" -s retraction_prime_speed="25" -s conical_overhang_enabled="False" -s draft_shield_enabled="False" -s wireframe_up_half_speed="0.3" -s speed_wall_0="30.0" -s speed_wall="30.0" -s sub_div_rad_add="0.4" -s raft_base_speed="22.5" -s support_offset="0" -s support_bottom_distance="0.1" -s jerk_roofing="20" -s wireframe_bottom_delay="0" -s gantry_height="55" -s meshfix_maximum_travel_resolution="0.8" -s material_bed_temperature="60" -s prime_tower_line_width="0.4" -s magic_fuzzy_skin_enabled="False" -s infill_offset_x="0" -s wireframe_height="3" -s bridge_wall_speed="15.0" -s raft_surface_line_width="0.4" -s layer_height_0="0.3" -s jerk_layer_0="20" -s bridge_skin_density_2="75" -s machine_nozzle_id="unknown" -s acceleration_print_layer_0="3000" -s default_material_bed_temperature="60" -s minimum_interface_area="1.0" -s max_skin_angle_for_expansion="90" -s wipe_repeat_count="5" -s infill_mesh="False" -s material_bed_temp_wait="True" -s bottom_layers="4" -s date="04-06-2019" -s wireframe_roof_outer_delay="0.2" -s top_skin_expand_distance="0.8" -s experimental="0" -s machine_heat_zone_length="16" -s support_zag_skip_count="8" -s cutting_mesh="False" -s skin_overlap_mm="0.02" -s mold_angle="40" -s speed_equalize_flow_enabled="False" -s spaghetti_max_infill_angle="10" -s time="17:49:20" -s minimum_roof_area="1.0" -s spaghetti_infill_stepped="True" -s prime_tower_size="20" -s switch_extruder_retraction_amount="16" -s machine_steps_per_mm_y="50" -s support_interface_extruder_nr="0" -s default_material_print_temperature="210" -s wireframe_fall_down="0.5" -s retraction_hop_only_when_collides="False" -s support_interface_offset="0.0" -s support_skip_some_zags="False" -s draft_shield_dist="10" -s raft_interface_fan_speed="0" -s layer_0_z_overlap="0.15" -s ironing_inset="0.2" -s flow_rate_max_extrusion_offset="0" -s acceleration_roofing="3000" -s raft_surface_jerk="20" -s limit_support_retractions="True" -s acceleration_travel="5000" -s multiple_mesh_overlap="0.15" -s gradual_support_infill_step_height="1" -s alternate_extra_perimeter="False" -s gradual_infill_step_height="1.5" -s machine_nozzle_temp_enabled="True" -s machine_show_variants="False" -s brim_outside_only="True" -s wall_x_extruder_nr="-1" -s travel_avoid_supports="False" -s prime_tower_flow="100" -s raft_interface_thickness="0.30000000000000004" -s bridge_skin_material_flow="60" -s speed_travel_layer_0="60.0" -s cool_fan_speed_min="100.0" -s prime_tower_circular="True" -s jerk_support_infill="20" -s machine_head_with_fans_polygon="[[-75, 35], [-75, -18], [18, 35], [18, -18]]" -s wipe_move_distance="20" -s bridge_wall_coast="100" -s bridge_skin_density="100" -s support_tree_angle="40" -s print_sequence="all_at_once" -s support_use_towers="True" -s wireframe_roof_fall_down="2" -s wipe_retraction_amount="1" -s jerk_enabled="False" -s bridge_wall_material_flow="50" -s raft_base_fan_speed="0" -s wireframe_printspeed_flat="5" -s bridge_skin_material_flow_3="110" -s cool_fan_enabled="True" -s clean_between_layers="False" -s travel_avoid_other_parts="True" -s support_pattern="zigzag" -s wipe_hop_amount="1" -s adhesion_extruder_nr="0" -s jerk_print_layer_0="20" -s material_print_temp_prepend="True" -s raft_base_thickness="0.36" -s wipe_retraction_retract_speed="25" -s infill_before_walls="True" -s support_type="everywhere" -s skin_angles="[ ]" -s jerk_infill="20" -s machine_max_feedrate_x="299792458000" -s support_join_distance= [... cut ...]
2019-06-04 17:49:28,106 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [171]: Socket state changed to Listening
```
I assume the "std::bad_alloc" is related to the slicing engine running out of memory. Is there a memory limit? The machine has 16GB of RAM and is not reaching this when the crash occurs.

**Expected results**
Slice should complete as normal.

**Additional Information**
No error message pops up in the CURA UI. It just shows the slicing progress bar (not progressing) until the user clicks cancel. If it is helpful I can probably add a project file to reproduce this to the issue.

smartavionics, 2019-06-04:
Hi @timhunderwood , please save the project (File -> Save) and zip up or rename the resulting .3mf file to be a .zip and attach to this issue. Thanks.

timhunderwood, 2019-06-04:
Thanks @smartavionics , zipped project file attached

smartavionics, 2019-06-04:
Sorry, you have to attach the file here in this box on the github page, you can't attach it to an email reply.

timhunderwood, 2019-06-04:
I attached it to the issue (at the bottom under Additional information). Is that not correct?

smartavionics, 2019-06-04:
Sorry, yes, it's there, I assumed it would be in the most recent reply, not the original message. My mistake.

I'm finished for the day, will look at it tomorrow morning. Cheers!

timhunderwood, 2019-06-04:
sounds good, thanks for your help! Let me know if you need more info from me.

smartavionics, 2019-06-04:
Actually, I took a quick look, it's the model that is the problem...
Install the mesh tools plugin from the marketplace and you can see for yourself.

[image]

Even so, it would be nicer if Cura failed more gracefully when presented with a bad model.

timhunderwood, 2019-06-05:
@smartavionics thanks for looking at this. I have installed the mesh tools plugin and I get the same summary message. 

However, when I generate a similar stl file that is much smaller (fewer vertices), I get the exact same error message and it slices (and prints) fine - (project attached)
[UMO_small_example_steps.zip](https://github.com/Ultimaker/Cura/files/3256218/UMO_small_example_steps.zip)

So it doesn't seem like this message is the direct cause of what goes wrong in the slicing engine.

I will try and look into the cause of the mesh tools error message. I agree that it would be great if Cura reported the error more gracefully as well.

Thanks!

Ghostkeeper, 2019-06-06:
> Is there a memory limit? The machine has 16GB of RAM and is not reaching this when the crash occurs.

There is no memory limit built-in in Cura, only from your operating system. But it could be that it tries to suddenly allocate an exobyte of memory at once. The slicer will crash before it even starts allocating any of that then.

Block137, 2019-06-06:
When I was slicing a model with 1,227,729 vertices, 117 MB file, Cura used about 3 GB of RAM.
I think I have sliced 300 MB file on PC with 8 GB RAM successfully before.

Ghostkeeper, 2019-06-06:
I ran your file through Cura while running CuraEngine through GDB. It took a while to slice (1600s) but that is mostly because it was running through GDB. But it sliced fine. Printing time: 40 minutes. My laptop is an old MacBook Air running Linux, 8GB RAM.

I don't know what caused it to take so long though, and why apparently it took so much memory for you. It's also pretty hard to debug now since I get a different result.

timhunderwood, 2019-06-06:
@Ghostkeeper thanks for trying this. It is definitely reproducible on my machine. Could be Windows vs Linux? Could you direct me to some instructions for running through GDB? I can try this and see if there is  more detailed error traceback. I can also see if I can try and run this on a Linux machine tomorrow and see if there is any difference

Ghostkeeper, 2019-06-07:
Sure. GDB is a debugger. For Windows, it's easiest to use the one that comes with MinGW: https://sourceforge.net/projects/mingw/files/

To run CuraEngine through GDB you'd call this first in a terminal:

    Cura.exe --external-backend

Then load your model and start a slice. It'll wait indefinitely at 0% now until you start CuraEngine through GDB:

    gdb --run CuraEngine.exe connect 127.0.0.1:49674

At some point this should then pause CuraEngine. You need to obtain a backtrace in GDB's command line then using the `bt` command. And perhaps some variables are worth looking into on the stack as well.

timhunderwood, 2019-06-07:
Thanks @Ghostkeeper ! I just ran it through gdb and I get the same error without much additional insight...

```
G90 ;absolute positioning" -s skirt_brim_line_width="0.4" -s acceleration_wall_0="3000" -s jerk_travel_layer_0="30.0" -s support_bottom_extruder_nr="0" -s min_skin_width_for_expansion="4.898587196589413e-17" -s extruders_enabled_count="1" -s brim_replaces_support="True" -s raft_airgap="0.3" -s cool_fan_speed_0="0" -s cool_fan_full_layer="2" -s alternate_carve_order="True" -s spaghetti_inset="0.2" -s cross_infill_density_image="" -s skin_overlap="5" -s max_extrusion_before_wipe="10" -s support_tower_diameter="3.0" -s retraction_hop="1" -s machine_max_feedrate_z="30" -s raft_base_fan_speed="0" -s wireframe_printspeed_flat="5" -s top_skin_preshrink="0.8" -s extruder_prime_pos_abs="False" -s speed_roofing="30.0" -s wipe_retraction_enable="True" -s material_standby_temperature="150" -s bridge_wall_min_length="5" -s retraction_combing="all" -s raft_jerk="20" -s machine_steps_per_mm_z="50" -s wireframe_top_jump="0.6" -s acceleration_wall="3000" -s coasting_enable="False" -s wireframe_printspeed_bottom="5" -s ooze_shield_enabled="False" -s wall_thickness="0.8" -s machine_height="200" -s min_infill_area="0" -s raft_base_thickness="0.36" -s support_interface_height="1" -s initial_layer_line_width_factor="100.0" -s support_xy_distance="0.7" -s acceleration_topbottom="3000" -s material_surface_energy="100" -s raft_surface_speed="30.0" -s z_seam_x="102.5" -s prime_tower_position_y="175.6" -s wireframe_printspeed_down="5" -s support_mesh_drop_down="True" -s cool_fan_speed_max="100.0" -s raft_surface_layers="2" -s prime_tower_min_volume="6" -s bridge_skin_material_flow="60" -s speed_travel_layer_0="60.0" -s top_bottom_pattern="lines" -s machine_nozzle_expansion_angle="45" -s retraction_combing_max_distance="0" -s material_bed_temp_prepend="True" -s support="0" -s material_flow_dependent_temperature="False" -s mesh_position_y="0" -s support_interface_skip_height="0.3" -s wall_min_flow_retract="False" -s expand_skins_expand_distance="0.8" -s support_tree_branch_distance="1" -s adhesion_extruder_nr="0" -s cross_support_density_image="" -s roofing_extruder_nr="-1" -s meshfix="0" -s infill_support_enabled="False" -s machine_shape="rectangular" -s prime_tower_flow="100" -s raft_interface_thickness="0.30000000000000004" -s machine_head_polygon="[[-1, 1], [-1, -1], [1, -1], [1, 1]]" -s draft_shield_height_limitation="full" -s machine_endstop_positive_direction_x="False" -s raft_interface_acceleration="3000" -s platform_adhesion="0" -s speed_print_layer_0="30.0" -s wall_line_width_0="0.4" -s machine_max_jerk_xy="20.0" -s magic_mesh_surface_mode="normal" -s infill_mesh_order="0" -s bridge_fan_speed_2="0" -s z_seam_y="615" -s machine_endstop_positive_direction_z="True" -s speed_prime_tower="60" -s wireframe_roof_inset="3" -s cool_min_layer_time="5" -s max_feedrate_z_override="0" -s wipe_retraction_extra_prime_amount="0" -s top_bottom_thickness="0.8" -s coasting_min_volume="0.8" -s speed_travel="120" -s cool_fan_speed="100.0" -s wall_overhang_angle="90" -s retraction_amount="6.5" -s machine_max_jerk_z="0.4" -s z_seam_corner="z_seam_corner_inner" -e0 -s cool_min_layer_time_fan_speed_max="10" -s retraction_min_travel="0.8" -s machine_settings="0" -s raft_interface_jerk="20" -s machine_extruder_end_code="" -s mold_angle="40" -s support_angle="50" -s prime_tower_wipe_enabled="True" -s support_roof_line_width="0.4" -s wipe_hop_enable="True" -s skin_preshrink="0.8" -s skirt_brim_minimal_length="250" -s support_line_distance="2.6666666666666665" -s line_width="0.4" -s support_conical_angle="30" -s speed_wall_x="60.0" -s speed_infill="60" -s wipe_repeat_count="5" -s support_fan_enable="False" -s top_thickness="0.8" -s jerk_travel_layer_0="30.0" -s multiple_mesh_overlap="0.15" -s limit_support_retractions="True" -s acceleration_travel="5000" -s raft_surface_jerk="20" -s bottom_layers="4" -s machine_nozzle_tip_outer_diameter="1" -s jerk_wall="20" -s spaghetti_max_infill_angle="10" -s support_tower_roof_angle="65" -s gradual_infill_steps="0" -s acceleration_ironing="3000" -s extruder_prime_pos_z="0" -s speed_support_bottom="40.0" -s travel_avoid_supports="False" -s acceleration_layer_0="3000" -s support_skip_zag_per_mm="20" -s travel_avoid_other_parts="True" -s machine_extruder_end_pos_x="0" -s experimental="0" -s top_skin_expand_distance="0.8" -s brim_width="8.0" -s speed_support="60" -s support_tree_branch_diameter="2" -s meshfix_maximum_deviation="0.05" -s support_roof_line_distance="0.4" -s support_bottom_enable="False" -s bridge_skin_speed_3="15.0" -s support_tree_collision_resolution="0.2" -s machine_steps_per_mm_y="50" -s jerk_travel="30" -s jerk_roofing="20" -s machine_extruder_start_code="" -s support_bottom_distance="0.1" -s meshfix_maximum_travel_resolution="0.8" -s jerk_wall_x="20" -s acceleration_wall_x="3000" -s infill_overlap_mm="0.04" -s raft_interface_line_spacing="1.0" -s outer_inset_first="False" -s material_flow="100" -s meshfix_extensive_stitching="False" -s switch_extruder_retraction_amount="16" -s switch_extruder_retraction_speeds="20" -s retraction_hop_enabled="False" -s prime_tower_line_width="0.4" -s machine_feeder_wheel_diameter="10.0" -s infill_line_width="0.4" -s fill_outline_gaps="False" -s ironing_flow="10.0" -s machine_min_cool_heat_time_window="50.0" -s spaghetti_infill_extra_volume="0" -s cool_fan_speed_min="100.0" -s retraction_extrusion_window="6.5" -s machine_steps_per_mm_x="50" -s mater [... cut ...]
[Thread 9220.0x2b58 exited with code 3]
[Thread 9220.0x22e4 exited with code 3]
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
[Inferior 1 (process 9220) exited with code 03]
```
For gdb I used the command:
```gdb -ex=r --args CuraEngine.exe connect 127.0.0.1:49674```

Later today, I should be able to try on a similar windows machine with more RAM (32GB) and see if that makes a difference. I can also try and run it on MAC OS... Maybe it is Windows specific.

Ghostkeeper, 2019-06-07:
What does the `bt` command give you? Is there a backtrace of where it terminated?

timhunderwood, 2019-06-07:
```
[Thread 10188.0x29bc exited with code 3]
[Inferior 1 (process 10188) exited with code 03]
(gdb) bt
No stack.
```
Let me know if I should try something else.

timhunderwood, 2019-06-07:
It just sliced fine on MAC OS (albeit a bit slow) - seems like a Windows only issue? Would be good if someone can confirm on a different Windows machine.

timhunderwood, 2019-06-07:
I have now tried running a larger version of the file on Linux (Ubuntu) and I am able to reproduce the error. On this machine I was able to slice the original example file without issue. It seems like the problem is related to memory allocation and occurs for smaller stl files on Windows.

The attached file below causes the error for me on Ubuntu with 32GB of RAM. Error in the log is:

```
2019-06-07 17:57:17,327 - DEBUG - [Thread-13] UM.Backend.Backend._backendLog [101]: [Backend] terminate called after thro$
2019-06-07 17:57:17,328 - DEBUG - [Thread-13] UM.Backend.Backend._backendLog [101]: [Backend] what():  std::bad_alloc
2019-06-07 17:57:17,588 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [806]: Backend quit wit$
2019-06-07 17:57:18,140 - INFO - [MainThread] UM.Backend.Backend._onSocketError [199]: Backend crashed or closed.

```

It would be interesting to see if you can reproduce the issue with the attached file.

[UMO_ubuntu_error_example.zip](https://github.com/Ultimaker/Cura/files/3267120/UMO_ubuntu_error_example.zip)

timhunderwood, 2019-06-09:
@Ghostkeeper I think the "platform: Windows" label may not be accurate - I don't think I have permissions to add/remove labels

Ghostkeeper, 2019-06-11:
Yeah that file causes my computer to freeze. Problem is that it freezes the computer before it can report the stack trace to me :stuck_out_tongue:

rburema, 2019-06-11:
(Pardon me for sneaking in like this...)

> So it doesn't seem like this message is the direct cause of what goes wrong in the slicing engine.

But you still get that for each model that has the problem right? I think its far to early to draw this conclusion.

That message has indeed nothing to do with the size _directly_. But it _does_ mean that the code runs on some wrong assumptions (you could make a good case that it still shouln't crash in this way then, but that's a bit out of scope of the current discussion), which could eventually lead to this behaviour.

Wrong assumptions basically means undefined behaviour, and when you get undefined behaviour all bets are off.

timhunderwood, 2019-06-11:
@Ghostkeeper / @rburema thanks for looking at this - at least we can now reproduce the issue!

@rburema - I agree that it is this type of file that causes the issue. All I meant by the above is that it is possible for the mesh message/warning to occur and for the engine not crash like this. 

Most importantly, I think the engine should at least not crash like this and display an error on the front end.

For context, the file is just a large number of cuboids that are all adjacent to each other. I'm not actually sure if this should be able to slice without issue or if the STL file should only define the bounding surface (with no internal surfaces)? The files get generated by some code I am working with. I am also working on adapting the code to produce just the surface :)

Ghostkeeper, 2019-06-12:
Internal surfaces tend to confuse CuraEngine more than necessary.

The part that it's getting stuck at is the slicing phase, which turns the model into slices. It's the first real phase of the process and normally takes a fraction of a second, but for this model it gets stuck. Each triangle gets intersected with a plane, resulting in a [line segment](https://github.com/Ultimaker/CuraEngine/blob/master/docs/slicing.md#triangles-to-lines). These line segments need to get [stitched](https://github.com/Ultimaker/CuraEngine/blob/master/docs/slicing.md#stitching) together to form polygons, which is typically the part that goes wrong when there are internal surfaces.

Ghostkeeper, 2019-06-12:
I removed internal surfaces with MeshLab. This reduced file size from 40MB (as binary STL) to 30MB, but didn't help much for the slicing time and memory usage in Cura.

rburema, 2019-06-12:
I'm sorry @timhunderwood, but we've decided to defer this for now. Though we agree that the error reporting from the engine to the front-end should be improved at some point.

timhunderwood, 2019-06-12:
@rburema no problem - I'm going to see if I can write some code to generate the file without internal surfaces (might take some time). I'll let you know if I still get any mesh warnings and whether it still causes any issues with the engine.

Thanks for looking into it!

timhunderwood, 2019-06-16:
@rburema @Ghostkeeper I managed to change the code produce the shapes without any internal surfaces. I now don't get the above warning when I run "Mesh tools" on the object before slicing. The file size is now much smaller (1-2MB) and I have made two files. The smaller one (50 x 50 array) slices correctly, the larger (100 x 100 array) still causes the bad alloc error. Both are attached below.

Understand if the issue is deferred but thought this might be useful for reference.
Thanks for your help looking into this.
[UMO_50_50_alternating_slices_correctly.zip](https://github.com/Ultimaker/Cura/files/3294245/UMO_50_50_alternating_slices_correctly.zip)
[UMO_100_100_alternating_slicing_causes_bad_alloc.zip](https://github.com/Ultimaker/Cura/files/3294246/UMO_100_100_alternating_slicing_causes_bad_alloc.zip)

closed 2023-06-24 by github-actions[bot]