Ultimaker/Cura #5853: [Feature request] Please add option to move Z axe before X/Y for next layer

neworld, 2019-06-05:
**Is your feature request related to a problem? Please describe.**

I am using Creality CR10s with original software. This printer knocks-off small parts quite often. I found, `G0` command behaves strangely on my print. It moves X/Y axis first and then Z axis. 

**Describe the solution you'd like**

Let's assume we have:
```
;LAYER:14
G0 X122.682 Y123.877 Z2.66
;TYPE:SUPPORT
```

I would like to be able to enable `move Z axis first for next layer` mode, which generates the following gcode:
```
;LAYER:14
G0 Z2.66
G0 X122.682 Y123.877 
;TYPE:SUPPORT
```

It ensures the printer head is above the last layer.

**Describe alternatives you've considered**

Use z-hop. It solves this problem but creates a stringing problem. Also, on some parts z-hop leaves a surplus of plastic. It grows and eventually, nozzle knocks-off this part as well. 

Also, I could write a script to post-process Gcode.

**Affected users and/or printers**
 

I believe it could help many users to reduce chances of knock-off parts without cons of z-hop

Ghostkeeper, 2019-06-07:
> It moves X/Y axis first and then Z axis.

No, it moves all 3 axes at the same time. At least, it should. And if there were an `E` in there it would move all 4 axes together. Most lines move the X, Y and E axes at the same time.

neworld, 2019-06-07:
> No, it moves all 3 axes at the same time. At least, it should. And if there were an `E` in there it would move all 4 axes together. Most lines move the X, Y and E axes at the same time.

It is what I am expecting. Also, G0 works without interpolation, so the Z-axis should move must faster than X/Y. However, by observing actual behavior seems like my printer move Z-axis at the very end of the movement.

Ghostkeeper, 2019-06-07:
Maybe some limited Z step size causing it to round Z coordinates?

neworld, 2019-06-07:
Maybe I don't understand what you mean, but printer moves to proper Z coordinates after knock-off print. Movement graph looks like:
```
    +
    |
X   |            +------------------------------------------------+
    |
    |
Y   |      +------------------------------------------------------+
    |
    |
Z   |                                                      +------+
    |
    |
    +----------------------------------------------------------------->
                                                                time

```

I will try first write transformation script and see if it helps.

LipuFei, 2019-06-17:
Fixed via https://github.com/Ultimaker/CuraEngine/pull/1076

closed 2019-06-17 by LipuFei

smartavionics, 2019-06-17:
Unfortunately, https://github.com/Ultimaker/CuraEngine/pull/1076, does not fix this issue because there is a travel move added to the end of the previous layer that uses the previous layer's z-height. See this g-code example...

```
G1 X2.664 Y5.656 E0.03292
G1 X1.939 Y5.943 E0.03242
G1 X1.179 Y6.139 E0.03263
G1 X0.397 Y6.238 E0.03277
G1 X-.387 Y6.24 E0.0326
;MESH:NONMESH
G0 F9000 X0.346 Y5.726       <---- travel move added at end of layer
;TIME_ELAPSED:38.204296
;LAYER:1
M140 S45
M106 S255
;MESH:zzz.stl
G0 F18000 Z0.4                  <---- the separated z and x/y travel moves you now output
G0 X0.346 Y5.726
;TYPE:WALL-OUTER
G1 F904.3 X1.061 Y5.638 E0.02995
G1 X1.76 Y5.46 E0.02999
G1 X2.43 Y5.197 E0.02992
G1 X3.063 Y4.851 E0.02999
G1 X3.646 Y4.428 E0.02995
G1 X4.172 Y3.936 E0.02994

```

smartavionics, 2019-06-17:
FYI, see `void LayerPlanBuffer::addConnectingTravelMove(LayerPlan* prev_layer, const LayerPlan* newest_layer)`

smartavionics, 2019-06-17:
It would be nice if the z could be increased before that connecting travel move is output but, of course, while processing one layer, you do not know the z-height of the next layer.

neworld, 2019-06-17:
What is the purpose of `connecting travel move`? I assume it is for cleaning hotend to avoid stringing/oozing.  If it moves a very short distance and moves always above print, then it is not a problem for knocking-off print at all.

smartavionics, 2019-06-17:
I don't know the purpose, it was added by one of the Cura devs. I will experiment with removing it and see what happens!

Ghostkeeper, 2019-06-17:
The connecting travel move makes it go to the starting position on the upcoming layer before the Z move and updates the state of being inside a mesh or not. It may be a long travel move; it's not guaranteed to be short. I was thinking that the implementation of this feature would go in that function.