Ultimaker/Cura #5854: [4.0.0] Per Model Settings: unexpected side effects on overlapping geo?

hooyah, 2019-06-05:
**Application Version**
Cura 4.0.0

**Platform**
Windows 10 x64

**Printer**
Ender3 (but I don't think that makes a difference)

**Steps to Reproduce**
1) import a model that has two overlapping meshes in it. (cylinder and cube) single stl.
[image]

2) slice -> the two meshes are "unioned" into one and the slices look as expected
[image]

3) import another mesh that overlaps any part of the first mesh(es). I'd like to use it as a cutter.
[image]

4) select the new mesh (sphere) and change the *Per Model Settings* to *modify settings for overlap with other models*. No settings have to be overridden at this point.

**Actual Results**
5) slice -> the internal overlaps of the first model (cylinder&cube) now create strange cavities and the two meshes will no longer print connected (see picture)
[image]

**Expected results**
5B) what I actually wanted to do was to add settings for *wall thickness*, *top/bottom thickness* and *infill density* and set them all to 0 in order to create a cutout shaped like the sphere. And this indeed works, except for the aforementioned anomaly 5).

Is this not how this feature is intended to work? I read that it used to be called 'cutting mesh' which I find incredibly useful, eg. 
to add optional cutouts to my prints.

[image]

**Additional Information**
Here's the Cura project (not actually a zip, just renamed)
[CE3_cubecyl.3mf.zip](https://github.com/Ultimaker/Cura/files/3256939/CE3_cubecyl.3mf.zip)

nallath, 2019-06-05:
No. That's not how it's intended to work.

You're trying to do solid modeling, which should be handled in the domain of CAD (Computer-aided design) software. Cura is CAM (Computer-aided manufacturing) software, so it's about how a model should be constructed (not how the model looks).

Cura expects a model without any internal geometry. It has some algorithms to handle the situations where this fails, but this is always a losing battle (as you can see from it failing in your case).

The cutting mesh is used to indicate that the overlap of the cutting mesh and the model need to be printed with another extruder (and not what you're trying to do: not printed at all. That should be handled when designing the model).

closed 2019-06-05 by nallath

hooyah, 2019-06-06:
Well if that's it's sole intended use then there are too many buttons that seemingly do the same thing.
Maybe you could write some documentation that explains how these options are meant to be used?
Why are these modes available at all if they are only meant for dual extrusion purposes when I only have one extruder installed?

nallath, 2019-06-06:
Well, the cutting can also be used to treat part of a model as a separate object. It's most common usage is to then print it with a different extruder, but there might be cases where you want to print both with the same extruder).

Ghostkeeper, 2019-06-07:
A cutting mesh is meant to cut out a piece of your mesh to print the intersection with different settings. Could even be a different extruder. It could set everything to 0 and literally replace part of your mesh with void.

That changes nothing in that your normal mesh has internal geometry which makes it inherently ambiguous what the inside is of your model. In your case you got lucky that it resolved the inside according to what you think the inside is. But when the sphere got involved you were not so lucky any more. It could be that involving another mesh would solve it.

To illustrate why it's ambiguous, here's an illustration of what the computer is thinking when it sees your model:

[image]