Ultimaker/Cura #5856: Cura 4.1 generates wrong gcode command causing the printer to stop printing

BenMens, 2019-06-05:
I have a stl that prints fine using Cura 4.0. When using Cura 4.1 the printer stops at layer 50. In the gcode I see a strange value that I marked bold below.

G1 **F17987547480000** Z5.2

More context:

G1 F3000 E2319.07248
G1 F17987547480000 Z6.1
;MESH:NONMESH
G0 F7200 X60.198 Y140.902 Z6.1
G0 X71.365 Y48.738
;TIME_ELAPSED:3294.172784
;LAYER:50
;MESH:corner.stl
G0 X71.365 Y48.738 Z6.2
;TYPE:FILL
_G1 F17987547480000 Z5.2_
G1 F3000 E2325.63484
G1 F4800 X72.446 Y49.819 E2325.66027
G1 X72.516 Y49.889
G0 F7200 X72.065 Y48.738
G1 F4800 X71.347 Y49.455 E2325.67714
G1 X71.277 Y49.525
G0 F7200 X76.641 Y49.819

rburema, 2019-06-06:
Hi @BenMens ,

We're aware of this issue. Printers that don't have a value set for maximum z speed get the maximum speed possible, light speed. Needless to say, this confuses some printers that didn't anticipate for relativistic speeds.

For now you can work around this by manually setting the maximum z speed or disable z hop or adjust the profile of your printer.

nallath, 2019-06-06:
This has been fixed for 4.2 (and it's a duplicate of https://github.com/Ultimaker/Cura/issues/5753)

closed 2019-06-06 by nallath