Ultimaker/Cura #5857: changing the way cura does concentric (bed adheasion, strength of part, etc.)

terramir, 2019-06-05:
Hi terramir here ;- )
Problem: some filaments (especially PETG) will stick to blue tape but smaller print moves glob together on tape if printed 1st other filaments can cause same problem.
Solution: print from the outside in this way the filament will stick on the outer lines and actually be able to complete the layer.
Also an option to only print the outer most layers concentric (with an option for the two outmost layers would be good) and print the other layers as lines preferably with alternating angle) this can give the parts more strength, also this can help prevent holes in the top layers due to insufficient infill support. 
Unless it's transparent filament a concentric pattern is only needed in the bottom and top layers maybe two layers (if printing really thin layers) also
Alternating the print even on concentric layers first from the outside in then the inside out because the lines are usually not exactly fitting i.e. there maybe a need for a fill line because the model is rarely exact multiples of line width in the total model means a slight offset of lines between the printing from the outside in and the inside out will improve strength and fill possible adhesion holes because of offset overlap between the 1st and second layer.
So request is 1st layer concentric pattern from the outside in (on the bottom) and the second layer from the inside out this should fix any bed stick and part holes that might occur.
also let's say you use 5 .2 layers after the second one a line pattern in alternating directions will create a stronger part and be invisible anyways.
Thanks for your time and this improvement would help several issues if it could be added to the features
terramir
terramir

Ghostkeeper, 2019-06-07:
> Solution: print from the outside in this way the filament will stick on the outer lines and actually be able to complete the layer.

In our experience, printing from the outside is detrimental for bed adhesion in all cases because it causes material to glob up in the centre, producing a large bump there which rips the print off the build plate. That's why it was [changed](https://github.com/Ultimaker/CuraEngine/commit/c86501427571f2d2879be7aa994e9b38615bd391) at some point.

> Also an option to only print the outer most layers concentric (with an option for the two outmost layers would be good) and print the other layers as lines preferably with alternating angle) this [...]

You can do this by setting the Initial Layer Bottom Pattern to concentric, the Top Surface Skin Layers to 1 and Top Surface Skin Pattern to Concentric. Lines and zigzag patterns always alternate direction.

So I think this is a "no". The ordering could be made configurable, but I don't think the use case for outside-to-inside is significant enough that people would actually be using that setting.

terramir, 2019-06-07:
ehh no you can't, the fixes you said do not slice only the top or bottom layer as concentric and the rest as lines or zig zag all top and bottom layers are concentric if you choose concentric I tried your fix and looked at the patterns in pronterface doesn't work
please reopen

Ghostkeeper, 2019-06-07:
Sure you can. Here is a project file and the resulting g-code as an example:

[top_bottom_concentric.zip](https://github.com/Ultimaker/Cura/files/3265172/top_bottom_concentric.zip)

GasAlbert, 2019-09-19:
> Sure you can. Here is a project file and the resulting g-code as an example:
> 
> [top_bottom_concentric.zip](https://github.com/Ultimaker/Cura/files/3265172/top_bottom_concentric.zip)

True, _you_ can. However, I must say it's kinda obscure the way you've got it working. I've been trying different setups with no results. I believe it's a common aestetical feature to have, so maybe making it more intuitive would be better for everybody.

I'm thinking on five comboboxes to choose first bottom, rest bottom, infill, top and last top layer patterns. Extreamly explicit and intuitive.

Ghostkeeper, 2019-09-23:
We literally have those comboboxes to choose the first bottom, rest bottom, infill, top and last top layer patterns. Only we have four, since we didn't find it useful enough to choose "rest bottom" and "top" patterns differently.

closed 2020-05-19 by Ghostkeeper