Ultimaker/Cura #5858: Snip-off sprue bed support between models

BoKKeR, 2019-06-05:
**Is your feature request related to a problem? Please describe.**

Bed adhesion can be problematic with multiple smaller parts. Take this example: [image]
I could apply brim that would make the parts connect but brim can be hard to remove. And its just waste of time and filament when this issue could be solved in a different way.
This is the resulted print fail:
[image]

**Describe the solution you'd like**

The solution I think would be the best is having a small tree type connection between the parts. It would help the parts stay as one united part. This support would be really easy to cut off compared to brim. This model of a reindeer has something similar modeled into the design. 
[image]

**Affected users and/or printers**
 
Anyone who prints smaller parts
**Additional context**

Is this possible using a custom plugin? The closest plugin that did something similar is this one: 
https://github.com/lokster/cura-custom-supports/tree/master/CustomSupports
But this is not an automated build process like brim/skirt/raft. This is support the user has to add manually to the build plate for each model. Can plugins be part of the slicing process like brim/skirt/raft and if so which type of plugin to build on? 

**Describe alternatives you've considered**

Having an edited cura-custom-supports plugin that would need manually added support between models.
Having an edited tree support plugin would be good enough too, that would work on the XY axis between the models instead of Z.

Ghostkeeper, 2019-06-07:
> Is this possible using a custom plugin?

You could make a plug-in that generates 3D geometry that joins them together. I don't think that is an ideal solution though. Normally I'd implement a feature like this in CuraEngine, however:

> Can plugins be part of the slicing process

No. Not at the moment.

> Having an edited tree support plugin would be good enough too, that would work on the XY axis between the models instead of Z.

The technical term for this is Minimum Spanning Tree. We have an implementation of this algorithm that works on single nodes (which is in fact used by tree support on a per-layer basis to determine in which direction the tree branches should join together). However in this case a tree is not ideal because it requires the maximum amount of tracing back and forth. In this case you'd need a TSP solver to find the most optimal route between all of these objects.

Ghostkeeper, 2019-06-07:
I'm wondering though if your solution would really make it more reliable. It could make it less reliable. Adding a sprue like this wouldn't make the individual pieces stick any better. It's only a millimetre of connection where it has extra surface area, so a few millimetres away from the sprue it won't have any beneficial effect on adhesion.

In fact, it could also ruin your print because if any part of the sprue connection lines get loose, it'll pull away all of your prints with it.

Ghostkeeper, 2019-06-07:
Perhaps a similar, but more reliable technique would be the old [fingers](https://github.com/Ultimaker/CuraEngine/issues/423) suggestion or the mouse ears. Still open, but deferred.

BoKKeR, 2019-06-09:
This would mostly help with smaller parts, When the extruder travels from one part to another part. The filament usually won't stick right away properly due to filament retraction/bed adhesion. It just curls up the filament into a little ball.

I added the snip off parts myself in blender and since then the parts printed without any issue while without it the part failed 5/5 times. 

[image]