Ultimaker/Cura #5859: Cura 4.1 can not connect to antcubic kossel

lofud, 2019-06-06:
**Application Version**
4.1

**Platform**
windows 10 64b

**Printer**
deltabot my printer is a anycubic kossel

**Actual Results**
wont communicate at all

**Additional Information**
worked fine in 4.0,  tried uninstalling 4.0 and 4.1 and deleting the application folders and printer driver and reinstalling 4.1 with no luck, but reinstall 4.0 and working fine again
[test.3mf.zip](https://github.com/Ultimaker/Cura/files/3259700/test.3mf.zip)

Ellecross, 2019-06-06:
hey @lofud did you use any plugin to connect to your printer? 
or do you mean 'connected' via USB?

lofud, 2019-06-06:
Yes via USB.
No plugins.

Ghostkeeper, 2019-06-07:
We are unable to test this because we don't have the Anycubic Kossel printer. All we can really do is rely on our technical community to figure out a way to fix this and apply the change (as long as it doesn't break it for the rest).

Ghostkeeper, 2019-06-07:
The only changes related to this that is new in 4.1 were [this commit](https://github.com/Ultimaker/Cura/commit/e56a97bfa2a06a561126fc26cc8c1967634ffe38) and [this commit](https://github.com/Ultimaker/Cura/commit/70e640be2276955f3befd72bba12538850ebd4ca). Perhaps they broke it for Anycubic? What exactly does the printer respond to a M105 command?

mbartlett21, 2019-06-08:
@Ghostkeeper 
My Anycubic Kossel Linear Plus is working fine.
I am using Cura 4.1.0 and used the deltabot profile like the instructions said to do.

Also, is the sd card inserted in the 3d printer? It does not show up if it isn't inserted.

ShikovYurii, 2019-06-11:
Confirm, can not connect to antcubic kossel, do not display temperature, do not move extruder, it see COM3, but dont  respond any command.

DmOcRsI, 2019-06-13:
I can confirm I am having the same issue with my CR-10S.  I have to shut down my computer all the way and then restart and then play a roulette of plugging and unplugging the USB cables before it connects correctly.  Cura is seeing the printer, but it's only displaying the extruder temperature and not the heat bed temperature and does not communicate with the printer.

Ghostkeeper, 2019-06-14:
If you connect to it with Pronterface and send it the `M105` command, how does the printer respond?

DmOcRsI, 2019-06-14:
Yes, Pronterface has no issues connecting at all.

paulgavrikov, 2019-06-16:
Can confirm that Kossel is not working after upgrading from 4.0.0-beta

```
>>> M105
SENDING:M105
ok T:20.0 /0.0 B:0.0 /0.0 @:0 B@:0
```

Ghostkeeper, 2019-06-17:
Hmm, that looks fine. It's more or less the same as what the Ultimaker printers respond with anyhow.

You guys got any log files of Cura?

sanczess, 2019-08-20:
I have the same issue with same printer and cura 4.2.1 sometimes
never had this with cura 4.0

KazeZlat, 2020-05-17:
Any progress on this? Still have to use 4.0 as of 4.6.1

KazeZlat, 2020-05-17:
As of logs it seems cura tries to detect baud rate and 4.0 detects 115200 after few retries. 4.6.1 thinks 250000 is the correct one.

ShikovYurii, 2020-05-17:
I fixed this problem myself, by modifying the script for searching for the maximum speed of the COM port, it sets too high a speed that the port emulator does not support over USB. I removed all speed options except for the 115200. this solved the problem. And they added higher speeds in versions with 4.0

KazeZlat, 2020-05-17:
@ShikovYurii do you have config entry name or how can we change this?

ShikovYurii, 2020-05-18:
Find 
C:\Program Files\Ultimaker Cura 4.5\plugins\USBPrinting\AutoDetectBaudJob.py

in file 

`# Copyright (c) 2018 Ultimaker B.V.
# Cura is released under the terms of the LGPLv3 or higher.

from UM.Job import Job
from UM.Logger import Logger

from .avr_isp import ispBase
from .avr_isp.stk500v2 import Stk500v2

from time import time, sleep
from serial import Serial, SerialException

#   An async job that attempts to find the correct baud rate for a USB printer.
#   It tries a pre-set list of baud rates. All these baud rates are validated by requesting the temperature a few times
#   and checking if the results make sense. If getResult() is not None, it was able to find a correct baud rate.
class AutoDetectBaudJob(Job):
    def __init__(self, serial_port: int) -> None:
        super().__init__()
        self._serial_port = serial_port
        self._all_baud_rates = [115200, _**250000, 500000, 230400, 57600, 38400, 19200, 9600**_]`

delete this

KazeZlat, 2020-05-18:
@ShikovYurii this requires users to build their own executables and might not be suitable for people with multiple printers.

ShikovYurii, 2020-05-18:
115200 most universal speed , and i'm user of cura, not developer. Its need only NotePad to do this

ShikovYurii, 2020-05-18:
Не знаю в чем по вашему проблема в конфигурациях с множеством принтеров, у читывая что вы сами прислали логи, предыдущей с версией и скоростью 115200. Эта проблема с китайскими конверторами сом- usb типа ch340 и тем что с версии 4.х+  они в этот скрипт дописали большие скорости типа 500000 которых ранее не было да и конектиться онотеперь быстрее

KazeZlat, 2020-05-18:
@ShikovYurii sorry, missed that you was writing about plugin which we can edit ourselves without building executables indeed. Your solution works for me, but some printers work better with higher speeds. 
While this is working workaround for those who has only one printer which is failing to monitor I think that correct solution will be config entry for each printer that will allow to override AutoDetectBaudJob behaviour.

Ghostkeeper, 2020-05-19:
Well, my expectation is actually that the user doesn't need to manually enter a baud rate. I'd expect that the AutoDetectBaudJob would automatically detect the baud rate. That's what it's named after.

This script tries to connect at various baud rates and sends `M105` commands to the printer, querying the temperature sensors on the printer. If the printer responds with a valid response, like `ok T:201 B:117` then it assumes that this communication is working.

So my next question is why the Anycubic Kossel responds with a valid temperature response on baud rate 250000, but doesn't allow printing at that baud rate?

ShikovYurii, 2020-05-19:
No, printer not respond absolutely.

Ghostkeeper, 2020-05-20:
In any case, we can't accept the proposed patch into the mainline code here because that would break the connectivity of other printers who are using 250kHz. Can we distinguish a correct response to the `M105` from the driver's incorrect response?

ShikovYurii, 2020-05-20:
as you can see in the message [ ] from the logs that, the program version 4.6.1 receives a response for the speed of 250K which immediately takes for correct and stops, but in version 4.0 continues checking and already selects the speed 115200 after a long check of all speeds, and selects the working speed. is it possible that the same check on the M105 command does not actually happen at all ?

KazeZlat, 2020-05-20:
@Ghostkeeper for me Pronterface isn't connecting at all. Might it be that 250k receives correct response bc 115k took time to connect and connected when 250k check happened?

I purpose to increase timeouts for first try or reorder and make 250k check first, fail it and then connect over 115k.

Ghostkeeper, 2020-05-25:
There is only one change that was made to USB printing since 4.0. It's this commit: https://github.com/Ultimaker/Cura/commit/70e640be2276955f3befd72bba12538850ebd4ca#diff-a0da3e9c4706ebfbfaa54c9c2db29ef8

You could perhaps go into the source code for the USB printing plug-in and remove those extra characters? See if that helps? The plug-in should be in your installation folder under the "plugins" folder.

Reordering is only going to move the problem to a different printer if that's an issue. I don't think that's any solution.

> Might it be that 250k receives correct response bc 115k took time to connect and connected when 250k check happened?

No, because the response is parsed at the same baud rate. So if Cura has already switched to 250k then it would double all of the bits that it received from the 115k call, and the response won't be correct any more.

closed 2020-06-19 by no-response[bot]

JerZeyHakx, 2020-08-16:
> Find
> C:\Program Files\Ultimaker Cura 4.5\plugins\USBPrinting\AutoDetectBaudJob.py
> 
> in file
> 
> `
> 
> ......
> 
> class AutoDetectBaudJob(Job):
> def **init**(self, serial_port: int) -> None:
> super().**init**()
> self._serial_port = serial_port
> self._all_baud_rates = [115200, _**250000, 500000, 230400, 57600, 38400, 19200, 9600**_]`
> 
> midify this
> ==>>> self._all_baud_rates = [115200]

_**### This fixed my issues with cura not responding via usb print**_