Ultimaker/Cura #5860: [4.1] Minimum Support Area: eating into parts

Block137, 2019-06-06:
**Application Version**
4.1

**Steps to Reproduce**
1.Set either "Minimum Support Interface Area" or Minimum "Support Area" to non-zero (larger than area of the part's slice)

**Actual Results**
Support interface (yellow) does not leave space for small feature of a part (red)
The portion ended up not being on the part and fall off or not extruded at all in real world.
Can occur for multiple layers until the area is large enough.

See left side:
[image]

**Expected results**
Support leaves holes for the part to be

**Additional Information**

This is 3 extruder setup
Nozzle 1: 0.25mm (red)
Nozzle 2:0.4mm (yellow)
Nozzle 3:0.5mm (white)

Vandrasc, 2020-01-16:
Hi @Block137 , would it be possible for you to confirm the fix in the latest master build?

Block137, 2020-01-16:
I don't have experience building Cura myself but I'll give this .3mf for you to test. Downward pointing horns are perfect for getting this bug.
Print settings are for testing only and are not practical at all.
[MinSupAreaTest.zip](https://github.com/Ultimaker/Cura/files/4072280/MinSupAreaTest.zip)

This file when sliced with 4.4.1
[image]

Vandrasc, 2020-01-17:
Hi @Block137 , thank you for your prompt reply. I would have sent you the build but sharing the project file with us helped a lot in confirming the fix, so thanks!

Both the support and infill leave space for small and thin parts of the model, the printing time did increase a bit, as expected, but nothing dramatic (2 minutes for your model) and the material usage decreased of course. 

See screenshot 
[image]

Also many thanks to @smartavionics for making a PR for this and fixing this kind of issues.

The fix will be in the next release.

closed 2020-01-17 by Vandrasc