Ultimaker/Cura #5861: Cura 4.1, does not execute extruder end gcode at end of print

wookie666, 2019-06-06:
**Application Version**
4.1
**Platform**
windows

**Printer**
custom fdm
**Actual Results**
extruder end gcode not placed at end of print
**Expected results**
end of print should be classified as end of extrder, and gcode should be implemented

**Additional Information**
This is my start and end gcode for head 1
[image]
This is my start and end gcode for head 2
[image]

I use a custom corexy printer with toolchange. 
When i start a print, it needs to pickup whichever tool it is using, which works correctly. 

When i end a print, it needs to put the tool away, but the extruder end gcode is not being implemented at the end of a print. I cannot use gcode end script, as i cant tell which tool is currently in the head, to know which tool can be put away. 
In s3d, i could use placeholders like, if currenttool=0 or currenttool=1. but cura cant do this. 

Basically. the extruder end gcode needs to be implemented at the end of  a print as well as tool change to another tool 

If i do a dual head print, the toolchange codes work fine, it will undock tool 0, print, dock tool 0, undock tool 1, print, dock tool 1, back to tool 0 etc. but just at the end, it will not dock whichever tool is in the head.
When i start a new print, i need it to pickup the new tool, which works fine, but the problem is, if it didn't dock tool 0 at the end of a print, if i start a new print with tool 1, it will try to pickup tool 1 with tool 0 still in the head.

rburema, 2019-06-06:
Hi @wookie666 ,

> In s3d, i could use placeholders like, if currenttool=0 or currenttool=1. but cura cant do this.

You can do things like `T{extruder_nr}` (or more specific things like `T{top_bottom_extruder_nr}`) though, which will evaluate to T0, T1, etc.

wookie666, 2019-06-06:
Can I use if statements?
How would I write a statement if t0 then run this code and if T1 then run
this code.

rburema, 2019-06-06:
Ah, I see the problem now, you'd need different paths for different tools.

I'll discuss this with the others.

wookie666, 2019-06-06:
Yes. Maybe even just a tickbox to be able to run the active tools end
script at the end of a print. That way it can be turned on or off. It will
solve the problem

Block137, 2019-06-06:
I'm also looking forward to this
Also this person at https://github.com/Ultimaker/Cura/issues/4909

Ghostkeeper, 2019-06-07:
One step further, you could also use `{machine_extruder_start_pos_x}` and `{machine_extruder_start_pos_y}` to move to the correct location in the end g-code.

Ghostkeeper, 2019-06-07:
We've discussed this in the team but decided that it's too much work for too little gain. The printers where this is necessary are rare, and we can't develop GUI elements for all minutiae without overloading users. Maybe we'll think of a solution some day, but we'll defer this for later work.

wookie666, 2019-06-07:
Wait a minute. You can't put a simple toggle to include the tool end gcode
at the end of a print or not because it will overload users?
Seems like a pretty necessary function if you ask me. The tool is finished
being used so the end code should run.

At least give us a way to add it into the end gcode and be able to tell
which tool is active using IF statements so that we can put the tools away.

Ghostkeeper, 2019-06-10:
> You can't put a simple toggle to include the tool end gcode at the end of a print or not because it will overload users?

No, I said it's too much work for too little gain and that the printers where this is necessary are rare.

I said that developing GUI elements for all minutiae [of printer configuration] would overload users. I understand that you think that your minutia is important, so if you'd like to do some of the work towards a solution, we'd love to see it. Otherwise you'll have to resort to modifying the configuration files manually in order to perform your if statements.

Note that the front-end doesn't currently know which extruder the print ends with. If you're going to implement this, it'll need a change in CuraEngine as well to send that over.

Block137, 2019-06-10:
Is there a reason why "Extruder End G-code" is not executed at the end?

I see no harm executing it at the end of every print, even on single extruder machines (they can be left blank).

IMO No need to make any GUI or user options, just always execute it.

wookie666, 2019-06-10:
This is what I've been saying.

Ghostkeeper, 2019-06-10:
Some printers need to do something with the extruder just before switching away from it that doesn't necessarily need to happen at the end of a print. Stuff like parking a print head, moving to a switching bay, wiping off the nozzle on a brush, etc.

wookie666, 2019-06-10:
No but it wouldn't hurt to happen at the end

Ghostkeeper, 2019-06-11:
If it doesn't hurt to happen at the end, you might as well paste the end g-code for all extruders in your printer's global end g-code (through the bracket syntax as above, or just copy-pasting the actual g-code). For some printers it does hurt.

wookie666, 2019-06-11:
You don't seem to get it. Each extruder has different park positions. How
hard is it to add the damn code. Fuck just tell me what I need to change in
the printer config at least.

rburema, 2019-06-11:
What @Ghostkeeper meant is _then just park **each** extruder_ at the end, so even when you only use the first extruder, also park the seconde one (in addition to the first), even if you don't use it. You can do that by adding _in the overall end code_ `{machine_extruder_end_code,0}\n{machine_extruder_end_code,1}` (or more if you have more extruders). This is also something you could add to the configuration.

If adding (an) extra extruder move(s) at the end doesn't hurt other people (as you say), then why would it suddenly hurt in your case?

Sorry, I know it must be very frustrating on your end, but please consider this:
Yes it's small, but you're not the only one asking for a change to that menu. And it _would_ eventually look like a parody of itself. Yes it's small, but we _literally_ have at least 1000 small(er) things that we could do that people are at least as passionate about. And _every_ one (or nearly so) of those changes has to go through review, component test, QA and sometimes system testing. Because it _will_ break something else (and maybe that something is someone elses printer...)

Ghostkeeper, 2019-06-11:
I do think I've understood what you're asking. You're just asking a whole slew of different things. Here are some requests that I can gather from this thread and their answers:

> end of print should be classified as end of extrder, and gcode should be implemented

No, it should not be put the extruder end g-code at the end of the print. That is the current expected behaviour.

> Can I use if statements?

Yes, but only when writing the .def.json file. Not from the interface.

> How would I write a statement if t0 then run this code and if T1 then run this code.

You currently can't, because the front-end doesn't know what extruder the print is going to end with. So even if you are writing the .def.json files with if statements, there is nothing your if statement can check on.

> Maybe even just a tickbox to be able to run the active tools end script at the end of a print. That way it can be turned on or off. It will solve the problem

That is a valid feature request, but we've deferred this because the use case is rare. We may implement it some day if we find that the priority increases. Or we may accept changes by someone else if that reduces the workload on the developers.

> Is there a reason why "Extruder End G-code" is not executed at the end?

Yes. Some printers break when an extruder end g-code is executed at the end of the print, for the same reason as when your printer breaks when all extruder end g-codes are executed.

wookie666, 2019-06-11:
I cant Park both heads because what happens, is

Head 1 is in use mounted.
Head 2 is parked already.

Head one then gets parked.
IF head two then tries to get parked, the main holder will crash into the
parked head 2.
Hence why i need to know {activeextruder} or something like that.

There has to be a way i can add if statements to the end gcode to check
which head is active and then run that code only.
something like if activetool = t0 {G1 Xxxx Yxxx Zxxx}

Is this at all possible

wookie666, 2019-06-11:
You keep telling me to put the extruder end code at the end print code. You
dont seem to understand, that they park at difference locations, If i try
to park a tool that is already parked, or try to pickup a tool that is
already picked up, there will be a crash

I have already implemented new gcode commands into marlin, where i place a
endstop in the toolholder that is triggered if the tool is loaded in the
tool bay. This way when i implement G6 T0, or G7 T0. it docks or undocks
T0.

Marlin then handles the if statement to either run the code or skip it
depending on the state of the endstop.
The issue is, i did this for a friend, and i use Smoothie, and have since
not be able to figure out how to add the same functionality to smoothie.

Basically, you'r saying that its too difficult to add a damn tick box,
which would solve every issue, because use case is rare, which is the exact
reason why a tickbox would be perfect. Seems legit.
Thanks for nothing then.

Ghostkeeper, 2019-06-11:
> Basically, you'r saying that its too difficult to add a damn tick box, which would solve every issue

Adding a checkbox is not a problem. Adding a checkbox that has a certain functionality is, because it's work that we don't have time for and don't give high priority because of the limited use case. If you think it is simple, be my guest and show us what simple solution you can come up with. We're normally not against merging that sort of feature into the mainline Cura.

Block137, 2019-06-11:
I'm also using Smoothie, I made an executable python program which scan every .gcode files in the folder and add docking sequence accordingly.
I'll try to make it easy to understand and share it if you want.

To devs: E3D tool changer will start shipping within 2 months (according to E3D website). This printer has 4 tools. Expect a wave of requests similar to this one in the near future.

golfromeo-fr, 2019-09-24:
it is not because some do crap things that you should keep doing the same mistake again and again

golfromeo-fr, 2019-09-27:
the {machine_extruder_end_code, 0} could help but what is needed is {machine_extruder_end_code, active_extruder}

{machine_extruder_end_code, {extruder_nr}} ?

golfromeo-fr, 2019-09-27:
putting the following line in "END G-code" seems to work fine.
{machine_extruder_end_code, extruder_nr}

where can (I/we) document this? where is the place to put this useful code please?

if I filter the final part of the gcode to show the load/unload of tools, the last tool is unloaded as expected

Line 48811: ;BEGIN LOAD FILAMENT T0
Line 48819: ;*** BEGIN LOAD FILAMENT T0
Line 50462: ;BEGIN UNLOAD FILAMENT T0
Line 50477: ;*** BEGIN UNLOAD FILAMENT T0
Line 50480: ;BEGIN LOAD FILAMENT T1
Line 50488: ;*** BEGIN LOAD FILAMENT T1
Line 866861: ;BEGIN UNLOAD FILAMENT T1
Line 866876: ;*** BEGIN UNLOAD FILAMENT T1

The--Captain, 2021-02-13:
Sad to see this is still a thing.

Devs defending broken code is even more unfortunate.  golfromeo-fr is correct.  To execute the begin code and not the end code leaves the gcode in an inconsistent state. 

Did Ultimaker manufacture a printer that expects this broken behavior, that they are so unwilling to fix the code to work properly?  

I appreciate there is a workaround, but it frankly should be the default.

luckybooger215, 2023-02-19:
Anyone find a better way of doing this yet? I have an infinite color changer that of course works much differently than your tool changer does but it has the same issue where the end of extruder Gcode is not executed after the end of the print. This ends up with the last color of the print being still fed into the extruder which when a new print in a different color is started it feeds the filament into the already full tube. Everything Wookie said is right about it should be just a "Apply End of Currently Operated Extruder Gcode to end of print Gcode", as it is in the long run the easiest solution that can apply to the most printers. You cannot apply the end Gcode of all extruders as it will end up in the feeding system of filament pulling the different colors except the one you were just using way too far back which will mess up the next print. I don't understand what Ghostkeeper was saying to Golfromeo-fr in their thread Golf mentioned above because he comes an inch away from stating the solution to his own problem. All you need is for the Gcode that Applies to the extruder to be in the end of extruder Gcode and the Gcode that applies to chamber, bed, cooling, etc to be in the global end of Gcode. This would make having a checkbox for it almost unnecessary as there would be no reason to disable it as long as it is only the current extruder that applies to the end of the print. This also aligns completely with what golf said that you should not have a beginning code for your extruder ever without an end for the same extruder, most people get away with it because single material is all they have but the point of this message is to illustrate that now is the time to adapt to printers with potentially infinite colors.

GregValiant, 2024-11-21:
I'm cleaning house.
Is this still a problem with current Cura versions (5.8.0 and up)?  Can I close this?
(BTW I have a script in Little Utilities (not in Cura) that will add the ending gcode.  I can post it if someone needs it.)

closed 2024-12-07 by github-actions[bot]