Ultimaker/Cura #5865: Latest version pauses for 5 secs between retractions.

rickyelqasem, 2019-06-06:
**Application Version**

**Platform**

**Printer**

**Actual Results**

**Expected results**

**Additional Information**

rickyelqasem, 2019-06-06:
Also build time for project using the same profile on 4.0.0 and 4.1.0 differ. 4.0.0 is 12:45 and 4.1.0 is 16:45

rickyelqasem, 2019-06-06:
I found others on Facebook facing same issue.

Block137, 2019-06-06:
More info here>>> https://github.com/Ultimaker/Cura/issues/5753
*Fixed for 4.2

Workaround: Set Maximum Z Speed to non-zero (15 is default for Ender 3)
Found under "Speed" tab
[image]

Ghostkeeper, 2019-06-07:
Duplicate of https://github.com/Ultimaker/Cura/issues/5753.

closed 2019-06-07 by Ghostkeeper