Ultimaker/Cura #5870: Cura 4.0 GUI

brazda, 2019-06-07:
**Application Version**
Cure 4.0

**Platform**
Linux Arch

I am having issue with the GUI, I am running Arch Linux and did a system update which upgraded my cura from 3.6 to 4.0.  The new GUI is not displaying properly and plainly sucks as a user interface. I want to switch back to the old GUI but cannot figure out how. 

Currently the text overlays over each other and makes it unreadable. the setting box is so small its hard to quick scroll through the setting. 

[image]
 
[image]

brazda, 2019-06-07:
Ran into another issue after upgrading, Core dumps when trying to preview 

2019-06-06 23:38:59,338 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [61]: Updating QualityProfilesDropDownMenuModel.
2019-06-06 23:38:59,354 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._checkStack [140]: Need to check for errors again. Discard the current progress and reschedule a check.
2019-06-06 23:38:59,469 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [136]: New error check scheduled.
2019-06-06 23:39:02,392 - DEBUG - [MainThread] UM.Controller.setActiveStage [159]: Setting active stage to PreviewStage
2019-06-06 23:39:02,427 - DEBUG - [MainThread] UM.Controller.setActiveView [92]: Setting active view to SimulationView
2019-06-06 23:39:02,430 - INFO - [MainThread] STLReader.STLReader.load_file [54]: Using legacy code to load STL data.
2019-06-06 23:39:02,483 - DEBUG - [MainThread] UM.Mesh.MeshData.calculateNormalsFromVertices [444]: Calculating normals took 0.0006566047668457031 seconds
2019-06-06 23:39:02,502 - DEBUG - [MainThread] UM.Mesh.MeshData.approximateConvexHull [401]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.01826333999633789 seconds. 351 input vertices. 351 output vertices.
2019-06-06 23:39:02,503 - DEBUG - [MainThread] STLReader.STLReader._read [97]: Loaded a mesh with 12612 vertices
2019-06-06 23:39:02,507 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/usr/lib/cura/plugins/SimulationView/simulationview_composite.shader]...
Segmentation fault (core dumped)

stelgenhof, 2019-06-07:
How did you install Cura? Using pacman/yaourt or the AppImage?

brazda, 2019-06-07:
pacman,  I just downloaded and ran the 4,1 appimage and it seems like the font overlay issue went away as well as it is not core dumping from the appimage either. So I am guessing I am going to need to run the appimage for now on. 

Still not digging the new layout anyway to change back the the old layout?

brazda, 2019-06-07:
Just to show from appimage it's working
[image]

stelgenhof, 2019-06-07:
@brazda The Cura from the Arch Linux User Repository (i.e. via pacman) uses versions of 3rd party libraries that differ from the AppImage version (AppImage is an official release from Ultimaker).

Because of this you may experience differences in the rendering, or even other issues. 

Not sure what you would like to change about the layout, but the AppImage is the current UI (4.x).

brazda, 2019-06-07:
Understand thanks, 

As for what I would like to change is to restore the 3.6 UI I thought it was a plugin as I did it on my windows machine but don't look like a option on the linux appimage.

stelgenhof, 2019-06-07:
The Sidebar GUI is still available on Linux if you prefer the previous layout. I don't use it myself as I prefer the new GUI, however am able to install it.

brazda, 2019-06-07:
I have that installed does not appear to do anything

stelgenhof, 2019-06-07:
Hmm, I did notice the version of that plugin says "4.0.0" so it may be possible it is not yet compatible with 4.1

fieldOfView, 2019-06-07:
I am indeed still working on updating that plugin for Cura 4.1. It will have to be updated for every new version of Cura.

Ghostkeeper, 2019-06-07:
Duplicate of https://github.com/Ultimaker/Cura/issues/5487.

The problem is that the AUR package allows Cura to use a different version of Qt that it was not designed for. We don't know if it's the 5.12 version that has a bug or if the 5.10 version that Cura normally uses depends on a bug to make it display right. In any case, it seems that this specific version of Qt in AUR is not displaying this correctly.

We can't upgrade the actual release of Cura to 5.12 at the moment because that would drop support for old MacOS versions that are still in use quite a lot among our users. And we also don't want to maintain both.

closed 2019-06-07 by Ghostkeeper