Ultimaker/Cura #5871: 4.1 Does not export/import correctly Dual Extruder sessings of the second extruder.

Eyos, 2019-06-07:
Second extruder: some settings are exported/Imported, many settings are not.

Liger0, 2019-06-08:
What about filling in the template? They're there for a reason and those infos are probably meaningful.

Ghostkeeper, 2019-06-10:
Yeah we're going to need some reproduction steps. I tried these reproduction steps, but it's working fine:

1. Change infill density to 25%.
2. Change infill pattern to Concentric.
3. Change adhesion type to None.
4. On the second extruder, change infill density to 26% and infill pattern to Cubic Subdivison.
5. Save to a profile called "MultiExtruderExportTest".
6. Export that profile to "MultiExtruderImportTest.curaprofile".
7. Import that profile in Cura.
8. Switch to the newly imported profile "MultiExtruderImportTest".
9. Assert that infill density and pattern is 25%/Concentric on the first extruder and 26%/Cubic Subdivision on the second extruder, and that the adhesion type is None.

Vandrasc, 2019-06-24:
I am also not able to reproduce this, @Eyos filling in the template will help us with some additional info in reproducing the issue you encounter. If we cannot reproduce, we cannot fix :(

closed 2020-06-19 by no-response[bot]