Ultimaker/Cura #5876: [4.1.0] Cura is heating and driving wrong tool when using prime tower

Schleichmichl, 2019-06-08:
**Application Version**
4.1.0

**Platform**
Windows 10

**Printer**
Custom Printer RepRap Flavor Core XY with E3D Kraken

**Actual Results**
Reproduction steps:

-create printer with 4 extruders and reprap flavor
-disable first 2 extruders
-set extruder 3 to PVA
-set build in draft profile
-enable "generate support"
-enable prime tower
-insert stl (e.g. https://www.thingiverse.com/thing:1843576)
-choose extruder no. 4 to print object
-Slice and save file to disk

As soon as Prime tower is used, Cura is heating up and driving T0.
I don't know if replacing in a post step would be a workaround, as the tool offset is made by Cura.

;FLAVOR:RepRap
;TIME:2588
;Filament used: 0.136997m, 0m, 0.239883m, 0.488552m
;Layer height: 0.2
;MINX:31.9
;MINY:31.9
;MINZ:0.3
;MAXX:98.8
;MAXY:98.8
;MAXZ:18.9
;Generated with Cura_SteamEngine 4.1.0
T2
M104 S215
M104 T3 S175
M109 S215
M109 T3 S175
M82 ;absolute extrusion mode
G28 ;Home
G1 Z15.0 F6000 ;Move the platform down 15mm
;Prime the extruder
G92 E0
G1 F200 E3
G92 E0
M83 ;relative extrusion mode
G1 F1500 E-6.5
;LAYER_COUNT:94
;LAYER:0
M107
M104 T0 S175 <-------------------------------------
.
.
.
G0 F3600 X72.045 Y72.241
M104 T0 S200 <-------------------------------------
G1 F1800 X72.239 Y72.047 E0.00516
.
.
.
G1 F17987547480000 Z1.3
T0 <-------------------------------------
M109 S200
M104 T2 S175

**Expected results**
Cura should use T3, as it would do without prime tower.

**Additional Information**
Printing without prime tower makes it impossible for my setup to get a clean print. Most of the time I have to cancel the print as support filament is missing at important locations where layer adhesion will be weaked layer by layer and finally fails.

Hope that this can be fixed quit fast.

Ghostkeeper, 2019-06-10:
Duplicate of https://github.com/Ultimaker/Cura/issues/1936.

The prime tower was never tested for >2 extruders, because we don't have printers to test with that have that many extruders. The initial implementation of the prime tower was not even designed with >2 extruders in mind. The new implementation from Cura 3.5 onwards was designed with this in the back of the mind, but was never tested with it.

I suspect the reason why it's choosing T0 to print the prime tower with is because it's choosing the material with the greatest adhesion tendency to print the outermost shell of the prime tower. If there are ties, it'll just choose the first extruder. In your case, the materials probably all have the same adhesion tendency and so it'll choose the first extruder.

This is not a problem with 1 or 2 extruders, since the prime tower only activates at all if more than 1 extruder is used, and with 2 extruders, both extruders are certainly used.

closed 2019-06-10 by Ghostkeeper

Schleichmichl, 2019-06-10:
Sorry, i think there is a misunderstanding. I don't have more than 2 extruders activated. I was even surprised that cura activates a disabled extruder. It seems more than a hard-coded Tool 0 piece of code.
Thus, it shouldn't make any difference if T2 instead of T0 and T3 instead of T1 is generated later on. 
The algorithm should work identically.

I will upload some data to make it more visual.
[image]

[CFFFP_c88.zip](https://github.com/Ultimaker/Cura/files/3272901/CFFFP_c88.zip)

Ghostkeeper, 2019-06-10:
Yeah, no I understood. Instead of choosing the first extruder in case of a tie, it should choose the first extruder that was used.

Block137, 2019-06-11:
I'm not sure, try setting "Prime Tower Minimum Volume" of the unused tools to 0.
T0 will still be used to print tower base however.

@Ghostkeeper How to adjust Adhesion Tendency? It's hidden on mine.

Ghostkeeper, 2019-06-11:
Yeah adhesion tendency can only be modified in the text files, not from the GUI.

Schleichmichl, 2019-06-14:
I don't understand why this should be a duplicate and why is still closed. There is bug in the prime tower algorithm.  It is not about choosing the wrong extruder, it's about enabling a completely disabled tool!
 
Altough the prime tower was only implemented for a certain use case (2 extruders), it can't be very time consuming to fix it, nor is this issue here violating the 2 extruder use case. It's only 2 different extruders.
As this can be tested without having the actual hardware, this isn't an impediment either.

Ghostkeeper, 2019-06-17:
It's not closed. The ticket is still open: https://github.com/Ultimaker/Cura/issues/1936

There is indeed a bug in the prime tower algorithm: It doesn't work with a printer with more than 2 extruders. That's what this issue is considered a duplicate of.

Schleichmichl, 2019-06-18:
Glad to hear ;-)