Ultimaker/Cura #5877: 4.1 USB connected printer stalling randomly with heating elements on

Carl-jackslabs, 2019-06-08:
**Application Version**
4.1

**Platform**
Windows 10

**Printer**
DIY Printer, Mega2560 board

**Steps to Reproduce**
Connect via USB and start a small print (~20 mins) 

**Actual Results**
Print starts with temperatures as expected, then at seemingly random times print will stop. All temperatures will be maintained and printer hardware remains responsive (not via Cura)

Attempting to click anywhere in Cura window causes a fatal error.

**Expected results**
Print should either (ordered best to worst expectations)
+ Finish (best case)
+ Fail but remember the line number of the job upon failure so there is a chance the job can be resumed
+ Cura at least not crash
+ Cura closing COM ports gracefully or re-establish a connection with machine (current workaround is dis-plugging USB and restarting Cura, restart at least should not be necessary)

**Additional Information**
I can't find a log file for Cura. Also sometimes there is an error message, sometimes not.

Ghostkeeper, 2019-06-10:
If you change the Maximum Z Speed setting to 10mm/s, does that resolve the problem?

Carl-jackslabs, 2019-06-10:
Thanks for the suggestion! 
As the issue is intermittent I can't be certain but I've completed two prints in a row now without a stall which is a higher success rate than I have had yet. I will come back here if a similar crash happens again.
I suppose this could be closed then?

closed 2019-06-10 by Carl-jackslabs

Ghostkeeper, 2019-06-10:
Well, what I described is a workaround. This is a duplicate of issue https://github.com/Ultimaker/Cura/issues/5753 then which is fixed for 4.2.