Ultimaker/Cura #5878: [FR] Write/read object names in 3MF

paukstelis, 2019-06-09:
Currently, writing to 3MF files does not same the object name (this is the object's file name when imported into cura). This means that upon opening the 3MF file the objects are given generic names (now the name of the 3MF file). It would be nice to have the original file/object name stored in the 3MF and assigned in cura. This would help differentiate objects that may be similar when a 3MF file is opened.

smartavionics, 2019-06-09:
I wish you good luck with this as I have previously asked for the object file names to be saved in the 3mf so that when a project is saved and  later loaded, hitting F5 will reload the models. The cura devs refuse to entertain the idea.

paukstelis, 2019-06-09:
That's a little different. That would require the complete path to be saved, and if you open on another system (or operating system) there is no guarantee that even exists. Having the object names would at least let you search for the models on the second system.

Ghostkeeper, 2019-06-11:
We'll have to look into what the 3MF supports.

paukstelis, 2019-06-11:
You can always add a metadata type, I would guess:
https://github.com/3MFConsortium/spec_core/blob/master/3MF%20Core%20Specification.md#341-metadata
This seems to be how PrusaSlicer does this from the one example I checked:
'<metadata type="object" key="name" value="z-axis-top-mod.stl"/>'

Vandrasc, 2019-06-24:
Devs see Cura-6587 for our reference. 

We created a ticket for it and it is currently in our backlog.

fieldOfView, 2020-05-05:
This has been fixed by https://github.com/Ultimaker/Cura/pull/7310

closed 2020-05-07 by Ghostkeeper