Ultimaker/Cura #5879: [4.1] Maximum Resolution not working

raulttro, 2019-06-09:
**Application Version**
Cura 4.1

**Platform**
Windows 10 Professional 

**Printer**
Creality CR-10s and Creality Ender 5. 

**Actual Results**
The default value of the Maximum Resolution setting in Cura 4.1 is 0.5 In Cura 4 it is 0.01. By changing the setting to 0.01 there is no difference in preview after slicing. After slicing with Maximum Resolution at 0.5 in Cura 4.1 actually looks exactly the same as slicing the same file in Cura 4 with setting at 0.01. I've tried different values with no observed change (even getting rid of the orange highlight by using a MR above 0.02.

**Expected results**
Changing the Maximum Resolution should have an impact on the how the object is sliced

**Additional Information**
See screenshots.

[https://i.imgur.com/PBVGpFN.jpg](url)
[https://i.imgur.com/chl9l5w.jpg](url)
[https://i.imgur.com/D5wCKcl.jpg](url)

Ghostkeeper, 2019-06-10:
Maximum Resolution plays together with Maximum Deviation. If your Maximum Deviation setting is so low that it's not allowed to remove any vertices, then setting the maximum resolution very high will not have any impact on the print at all.

Try fiddling with Maximum Deviation as well. See if you get it to work then.

Otherwise, we're going to need to see a project file to see where the problem lies. It seems to work fine using the models and settings I'm testing with.

raulttro, 2019-06-22:
For some reason I didn't had that setting visible. In earlier versions of Cura I had them marked all as visible but it in 4.1 they weren't by default apparently. Thanks for the clarification!

closed 2019-06-22 by raulttro

Ghostkeeper, 2019-06-26:
Yeah a bit of a usability issue is that making all settings visible makes all current settings visible. It doesn't make all future settings visible as well.